kylecorry31/Trail-Sense #11: Allow manual calibration

kylecorry31, 2019-12-28:
- Declination
- Altitude

aw-bib, 2020-05-30:
Note that it might be easier to know the current sea level pressure at your location (e.g. by some official, calibrated measurement) then the correct altitude. So it would be great if one could use this value instead of altitude.

kylecorry31, 2020-05-30:
I think there will be a few settings around this, such as when you enter pressure/altitude you can set it as constant (altitude), dynamic (sets the altitude as a baseline, but it can change - need to come up with a better name), or automatic (uses GPS for altitude)

kylecorry31, 2020-05-30:
This should clear up what I mean:

**Settings for altitude**
(Disabled if weather monitoring if off)

- Initial altitude (Recorded once - at time of changing this setting, and used as a baseline)
    - Manual
        - Enter sea level pressure
        - Enter elevation
    - Automatic
        - GPS (Default)
        - Barometer
- Auto calibration
    - Off (does not use GPS/barometer and will keep altitude constant to what was entered above)
    - On
        - GPS (Records GPS elevation in background - what currently happens) (Default)
        - Barometer (Uses barometer to adjust the initial altitude, a variant of the current algorithm which only factors in rapid barometric changes to avoid drift)

kylecorry31, 2020-05-30:
**Settings for declination**
- Initial declination
    - Manual
        - From lat/lng
        - From degrees
    - Automatic
- Auto calibration
    - Off (Will not use GPS to update declination)
    - On (Will use GPS to update declination)

kylecorry31, 2020-05-30:
**Settings for location**
- Location
    - Manual (from lat/lng - with button to use current GPS location, like the beacon screen)
    - Automatic

Note: the GPS location is only used on the navigation and astronomy tabs.

Setting GPS to manual will disable the beacons (would there be a reason why they should be enabled here? I can't envision that navigation without GPS would be useful since moving will change the direction to the beacons, and it wouldn't be reflected in the app).

Anything that uses the GPS location (ex. declination, astronomical times) will respect the value set here.

kylecorry31, 2020-05-30:
The auto-calibration values will be available directly on the settings screen, while the initial values will be set on a separate page accessible from the settings tab.

kylecorry31, 2020-05-30:
Cached properties needed:
- Last altitude (exists)
- Last location (exists)
- Last declination

(FYI, I'm putting some of this information here for me to refer back to when implementing this, but feel free to comment on any of it)

kylecorry31, 2020-05-30:
Flow for the altitude / declination will be:

1. Go to settings page
2. Scroll to calibration section
3. Click calibrate X (where X is altitude or declination)
4. Opens initial value page
5. Select from manual or automatic
6. Enter the initial value (ex. manual elevation or select GPS)
7. Select update policy (auto calibration)
8. Return to settings screen - you can see the last stored value (if update policy is fixed, else 'Auto updates') and the update policy, which can be adjusted directly on the settings screen

Flow for the GPS will be:

1. Go to settings page
2. Scroll to calibration section
3. Click GPS update policy
4. Opens initial value page
5. Select from fixed or automatic
6. If fixed is clicked, enter the current location (similar page to the drop beacon page)
7. Return to settings screen - you can see the last stored value (if update policy is fixed, else 'Auto updates')

swedneck, 2020-05-30:
Would it be possible to also manually calibrate the compass?
Sometimes it just thinks north is completely the wrong direction, and it would be nice to be able to force calibrate it using some other method of finding north.

Perhaps method 2 in this article could even be integrated for semi-automatic calibration: https://www.wikihow.com/Find-True-North-Without-a-Compass

kylecorry31, 2020-05-30:
Yeah,

I can do the compass calibration technique where you rotate your phone to calibrate.

Also, you will be able to use the declination calibration to adjust the compass of it is still inaccurate

kylecorry31, 2020-06-07:
FYI, this is a large feature and I don't have much time the next few weeks, so I am going to have to push the delivery of this further out

kylecorry31, 2020-08-29:
(not ready for release yet - hopefully next week)
[image]

(need to redesign this UI to match the altimeter)
[image]

kylecorry31, 2020-08-30:
Almost done with this - going to do some more testing tomorrow and then send out a new debug build, release likely next week - have a few more things I need to include in it.

aw-bib, 2020-08-30:
May I ask wrt altitude correction: as mentioned above from a practical point of view it might be easier to know the current sea level pressure from some calibrated official source. Would this end up in `set overwrite from barometer`? Also you mentioned in the discussion some table used to map GPS hights. This would end up in `set overwrite from GPS`?

kylecorry31, 2020-08-30:
So the seal level entry does come from set from barometer (I should probably rename that or add some text to avoid confusion)

The altitude correction does not get applied using the set from GPS - that falls under the set altitude correction setting and only applies when the altitude is set to auto. The offers only include the difference between GPS location and actually, so I can't use them as a lookup (though in the future I would love to use something like DEM to allow that)

kylecorry31, 2020-08-30:
I'm also going to change the elevation correction setting to be called something like convert elevation to MSL (mean sea level), which is actually what it's doing (GPS uses wgs84). I can make that setting not disabled when auto location is disabled so that the setting it factors it into the set override by GPS button.

aw-bib, 2020-08-31:
> I would love to use something like DEM to allow that - may be too big of a file though

I wonder if one could rely on files from other apps? E.g. Osmand should use this for hill shading. Similar argument would go for [OruxMaps](https://oruxmaps.com/cs/en/). Maybe one coIuld share the data file? _Use it if it exists anway_ kind of logic.

closed 2020-08-31 by kylecorry31