kylecorry31/Trail-Sense #39: Submit to F-Droid

roshavagarga, 2020-04-10:
If applicable, I'd be interested in seeing this app available on the main F-Droid repository :)

kylecorry31, 2020-04-10:
I plan to release it into Google Play Store once development is finished (pretty soon!). I'll look into F Droid as well, but most likely will only post it on Google Play

roshavagarga, 2020-04-10:
F-Droid is a non-google app repository that focuses on its apps being open source and not having trackers and non-foss parts embedded within them. Thank you for the reply, if you were to give permission to do so, the F-Droid developers could build the app from this source on their own with your permission if it doesn't have any of the abovementioned things.

kylecorry31, 2020-04-10:
Oh, I was not aware that the F-Droid developers would handle building / publishing of the app. In that case, I give permission to host on F-Droid. 

What would I need to do for it to be hosted there? Do you know how F-Droid handles versioning (does it just always use my latest master commit, or do I need to create releases)?

Also, there is probably another week or two worth of testing and bug fixing before this is ready for a production release (though I'm fine with it being released early on F-Droid)

roshavagarga, 2020-04-10:
You make a request for packaging [here](https://gitlab.com/fdroid/rfp/issues). The app should abide by the [inclusion policy](https://f-droid.org/en/docs/Inclusion_Policy/). The app should also use the [fastlane file structure](https://gitlab.com/snippets/1895688). There's a further how-to available [here](https://f-droid.org/en/docs/Inclusion_How-To/).

As far as your other question - versioning is handled by creating releases. Fastlane makes it easier to update some things like the description and changelog on F-Droid without actually making a release, so that's why they recommend using it.

Typically the process requires an automated bot and/or volunteer to browse through your code and make sure you aren't using anything proprietary that can't be built from source. After that, with or without fastlane, as long as it gets approved it gets added to the build queue and a page is automatically created. If you want auto-updating to be enabled for your app, you'll need to use static values for versionCode/versionName in your build.gradle. This process can take some time, typically people just submit their stuff and if they don't get a reply within 2 weeks they are free to directly ping one of the volunteers.

There are already a few compass apps but all of those are old and probably java-based, so yours would definitely be a welcome edition. The F-Droid ecosystem is a lot smaller and tends to be for the more privacy-oriented users, so I'm sure an open source Kotlin app like yours would be a welcome edition.

If you have any questions about whether something you already use in your app would be acceptable, feel free to ask - I'm not a developer, but I've browsed enough to be able to tell you what's acceptable and what isn't or at least know who to ping for a better reply :)

If you plan on adding analytics or advertisements - those are acceptable if you use FOSS parts to do so. For analytics the current option is ACRA. Unlike Google Play, you can also add a donation link which will pop up directly on your app's F-Droid page :)

kylecorry31, 2020-04-10:
Thank you for all the information!

I hope to be able to kick off this process tonight (maybe I can also use F-Droid as a way to get some initial feedback on this app)

kylecorry31, 2020-04-10:
Unfortunately, I think the inclusion policy would prevent this app from being on F-Droid due to its dependency on 'com.google.android.gms:play-services-location:17.0.0'

Although, I may actually switch to the default Android LocationManager since I kinda only want to use the GPS (and not WiFi or network based lookups)

roshavagarga, 2020-04-10:
I generally don't see much use for Wi-Fi assisted locations outside of populated areas, so it might be more meaningful to just use the default ones, yes :) Alternatively, if I'm not misunderstanding the issue - maybe add the option to make use of [this](https://f-droid.org/en/packages/com.google.android.gms/) with a proper backend as a fallback?

kylecorry31, 2020-04-10:
Thanks! I ended up just removing the Google Play Services in general and switched to the location manager (I'll need to do some testing to ensure everything still works properly before publishing it though). 

Other than that, everything looks like it should be ready to host on F-Droid. I'm going to have to postpone looking further into this tomorrow / creating the proposal.

roshavagarga, 2020-04-11:
Sounds great! As I said, the waiting period to get approved takes a while, so feel free to submit it. A lot of the apps that are on F-Droid don't reach their first stable version before being added, so it isn't much of an issue if you submit it earlier as long as you've done the fastlane thing. Once again - thank you for being open and considering this, it's always good to see quality open source applications being added to F-Droid :)

As a sidenote, since you mentioned feedback, please be aware that F-Droid does not have grading or comment section, but after you've published you're always free to promote the app through relevant subreddits - /r/androiddev, /r/fdroid, /r/fossdroid and /r/privacy come to mind. Besides these, I'd take a look at subreddits aimed at survivalists, mountaineers and hikers - an app such as yours seems like a good companion app for them, so their suggestions should be good feedback.

Do note that some of the above subreddits feature users that are very privacy-centered, so they would be more picky about permissions, having information on how they're used and whether the app shelters its information from other apps and/or sends data to an outer source. I know that there is an inherent flaw in how GPS is used - something about the information being sent between the device and the GPS not being secured by default, which isn't something you can fix as far as I'm aware. Maybe think about how the app does what it does and whether there's anything that needs hardening, as well as working on a bit of documentation about which permissions are used for what and a good description of the app itself.

Out of pure curiosity, does the use of the default Android LocationManager enable you to make a proper system for choosing the correct type of GPS or is that something that the LocationManager itself does on its own? My thought process is along the lines of users that live in areas which support numerous types of GPS and how your app would know which one to use - if their device supports it. Example: User is in Europe, your app identifies the presence of both GALILEO and standard GPS - which one gets chosen, since GALILEO should theoretically provide more accurate data when a stable connection and enough satellites are present.

kylecorry31, 2020-04-11:
Thanks for all the information!

The LocationManager chooses the correct positioning system itself, so I don't need to factor that in. Also, from initial testing, it actually appears that the LocationManager is more accurate than the Play Services version (probably because it isn't using network location).

As of right now, I am not using any maps, though I did experiment with osmdroid for mapping. I ended up deciding that I didn't want maps because I don't want this app to use any Internet data. I might create a companion app for downloading offline topographical maps though (I don't see too many good ones out that that don't require a subscription)

roshavagarga, 2020-04-11:
Maybe these? https://search.f-droid.org/?q=offline+map&lang=en

kylecorry31, 2020-04-11:
OSMAnd~ was exactly the kind of app I was looking for, thanks!

roshavagarga, 2020-04-11:
You can take a look at how Open Tracks implements it through a separate OSM dashboard (both available in F-Droid).

kylecorry31, 2020-04-11:
https://gitlab.com/fdroid/rfp/-/issues/1303

closed 2020-04-14 by kylecorry31

roshavagarga, 2020-05-14:
@kylecorry31 Trail sense is now officially [available](https://f-droid.org/en/packages/com.kylecorry.trail_sense/)

You can update the readme.md as can be seen [here] with the appropriate links for Trail Sense.(https://github.com/leoxshn/posidonLauncher/commit/297954290f655fb6a76d0431fa9d1eaf40d9d29f)

As was previously said, thank you tremendously for the work you've put in and for making this great app available! If you encounter any issues, feel free to post an issue on the F-Droid gitlab. Typically after you tag a release with an updated changelog it normally takes up to 7 days for it to reach the website, sometimes less, sometimes more. If it takes more than that there might be a build issue :)

kylecorry31, 2020-05-14:
Thanks for all the help getting this onto F-Droid! I'll add that link to the readme.