kylecorry31/Trail-Sense #52: Location permissions can be optional

dennisguse, 2020-05-14:
Compass and barometer do not require the location permission.
So, location permission should be made optional if the user needs the beacon feature.

kylecorry31, 2020-05-14:
Good point, thanks!

kylecorry31, 2020-05-15:
FYI, I committed this to master. Denying the location permission will have the following side effects

- The weather feature will still work, but you will not be able to convert it to sea level pressure (which will make the forecasts less accurate while moving between elevations). 
- Beacons are disabled
- The location displayed (if any) on the navigation screen will be the last location you were when Trail Sense had location permissions
- The astronomy tab may be inaccurate (same as above point with cached location)
- True North will not work
- Altimeter may be inaccurate (only uses barometer)

I didn't disable any of the settings when the permission is denied - that is a low priority, so I may revisit it later.

kylecorry31, 2020-05-15:
I should also mention that some of those issues can be remedied once I implement the manual calibration feature.

closed 2020-05-15 by kylecorry31