kylecorry31/Trail-Sense #54: How to quit Trail-Sense?

dennisguse, 2020-05-14:
I could not figure how to close trail-sense (also remove notification).
Did I miss something?

kylecorry31, 2020-05-14:
Hi,

Trail Sense will run in the background to monitor the weather (if it doesn't do that, it won't be able to give you weather forecasts or show the barometer history). Android requires that a notification is present for apps that run in the background like this - I can research this further, but I expect I won't find much of a solution here. 

What I did on my phone, was set the "Weather" notification category for Trail Sense to Silent and Minimize within Android's settings.

Would you expect to be able to opt-out of the weather monitoring feature? If so, I can add a setting for that.

dennisguse, 2020-05-14:
@kylecorry31 No, I meant that I want to shutdown the weather monitoring if I don't need it.
Like in home office ;)

kylecorry31, 2020-05-14:
Okay, I think I am going to create a setting for this to turn on / off the weather monitoring and allow it to be turned off via the notification. How does that sound?

Thanks for the feedback!

kylecorry31, 2020-05-14:
[image]

kylecorry31, 2020-05-14:
Or would you expect that after stopping the monitoring, you would need to take another action within Trail Sense to restart it, rather than just opening the app?

dennisguse, 2020-05-14:
Sounds like a great idea.
I like it :)

closed 2020-05-14 by kylecorry31