kylecorry31/Trail-Sense #55: Compass freaks out when moved

swedneck, 2020-05-14:
The compass works very well as long as the device is stationary, but as soon as it's moved the compass completely freaks out and becomes useless.
I presume this has something to do with the data not being smoothed.

kylecorry31, 2020-05-14:
That could be the issue - what model is your phone?

I am not experiencing this issue on my Pixel 3a, but some phones may have noisy accelerometers/magnetometers which could cause this issue (and hence I would need to add some smoothing there).

swedneck, 2020-05-14:
I have a lenovo p2a42

kylecorry31, 2020-05-14:
Thanks, I'll look into this more - I'm assuming other compass apps work fine on your phone, correct?

swedneck, 2020-05-14:
Sort of, other apps seem to have the opposite issue where they're very slow to find north or think north is in a completely wrong direction, but seeing as a dedicated sensor reading app gives me accurate magnetometer readings that just seems like the apps themselves having issues.
I haven't tried any nonfree apps though.

For what it's worth the trail-sense compass seems to be completely accurate once the phone is placed on a still surface.

kylecorry31, 2020-05-14:
Yeah, a lot of times it's a tradeoff between speed and noise. I'll see what I can do to solve this, though it may be tricky since it seems to work fine on my phone (and I'm the only developer).

swedneck, 2020-05-14:
Let me know if i can help in some way, like testing a debug build.

kylecorry31, 2020-05-14:
That would actually be super helpful! I'll post the debug build(s) here for you to download when I'm ready. 

Thanks!

kylecorry31, 2020-05-14:
@swedneck can you let me know if this is any better? 

https://github.com/kylecorry31/Trail-Sense/blob/master/temp_trail_sense.apk 

I added some smoothing to the compass data, hopefully this will resolve the issue.

swedneck, 2020-05-14:
That is waaaaaay better! A bit less filtering would be nice but if that's not possible then this is perfectly usable.

dennisguse, 2020-05-14:
@kylecorry31 I cannot recommend to put the apks into the repo as those binary files will just bloat the repo and will remain in the history forever. So, every clone will download it. 
You could put them in a branch and delete the branch when not needed - not nice but less bad :)

kylecorry31, 2020-05-14:
@dennisguse Yeah, I knew that I was doing a bad thing when I did it - I think I will go with the branch solution, I always thought GitHub allowed uploading apks to issues comments, but I guess not

kylecorry31, 2020-05-14:
@swedneck glad to hear! How about this one: https://github.com/kylecorry31/Trail-Sense/blob/testing/temp_trail_sense.apk

I reduced the filtering by 40%

swedneck, 2020-05-14:
hold on, why not just let the user set the smoothing manually?

swedneck, 2020-05-14:
fwiw i just tested it and that amount of smoothing is quite nice.

kylecorry31, 2020-05-14:
I added the smoothing as a preference, thanks for the great idea! You can access it on the settings tab: https://github.com/kylecorry31/Trail-Sense/blob/testing/temp_trail_sense.apk 

The smoothing you were using for the last apk was 30 on that slider. 

If this looks good to you, I will close out this issue and delete that apk branch :)

swedneck, 2020-05-14:
Lovely!

However it seems i have discovered 2 more problems!
By moving the phone in a tight circle i can somehow make the compass spin, i can even control the speed at which it spins, which is interesting.

I think this is related to how the compass seems to overshoot the rotation of the phone, if i quickly rotate the phone 90° the compass won't end up pointing the same direction as previously.

kylecorry31, 2020-05-14:
Thanks for testing that for me!

I'm not sure if I follow the first problem? 

From what I think you are saying is that if you move the phone in a tight circle, the compass spins, and I'm assuming it keeps spinning when you stop, right? Does it ever stop, or do you have to restart the app?

As per the second issue, I think I can reproduce that as well, though it is hard to tell because it is hard to line it up to where you were before. How far off is it for you?

I am actually using Android's orientation sensor for the compass (it is deprecated, but works much better than the solution they want developers to use now), so it could be that it is not keeping up with the movement with its sensor fusion algorithm.

swedneck, 2020-05-14:
https://ipfs.swedneck.xyz/ipfs/QmXwb9h35Lw2pyxJRphXsdqnysi4AwNHYHUwoRJBRn5HT9?filename=ScreenRec_20200515_010629.mp4

First i moved the phone in a circle, then i quickly rotated it about 90° left and right.

kylecorry31, 2020-05-15:
@swedneck Thanks, I'll look into that tomorrow and see if I can replicate it at all. Yeah, that looks like could be some accuracy issues with the orientation sensor. I can look into the recommended way of making a compass and seeing how that looks (maybe allow the option to choose between them, unless I get the new way working very well). Also, you may want to delete that link from your comment as it contains your location (unless it is a mock location).

swedneck, 2020-05-15:
no worries, i thought about the location being shown but concluded that i don't really care :P
But it doesn't hurt to remove it anyways.

kylecorry31, 2020-05-15:
Hi, I may have addressed the two issues you found - not completely sure, I was unable to reproduce the first one and was having some trouble with the second one. I added a setting to "use experimental compass" in the settings tab that you can turn on to test it out: https://github.com/kylecorry31/Trail-Sense/blob/testing/temp_trail_sense.apk 

For me, this setting adds a bit more noise and is slower than the regular compass (though, it is what Andoid is encouraging developers to switch to), but it may be more accurate when moving than the previous one.

swedneck, 2020-05-15:
Experimental compass seems to solve all the aforementioned issues, even removing the need for filtering. (however i see no need to remove that setting)

Dark theme looks great :D

kylecorry31, 2020-05-15:
Glad to hear! Interesting, the non-experimental one works best for me, I'll definitely leave that setting in though for phones like yours. 

Thanks for all the help testing this build, I'll submit a release to F-Droid pretty soon.

closed 2020-05-15 by kylecorry31

swedneck, 2020-05-15:
Maybe just rename the option? "Experimental" is likely to dissuade people from using it, maybe "Alternative compass API" or something?

kylecorry31, 2020-05-15:
Agreed, for now I am going to keep it as experimental until I do more testing of it (I do want to get it out to everyone though). Then I'll likely rebrand it  - I'm going to think more about the name, I don't want it to be too technical.

kylecorry31, 2020-05-15:
What I might do, is with enough testing, I may make this the default option and rename the setting to "Use legacy compass" or something like that. Especially since the original compass algorithm I was using is depricated (and has been since some early Android versions)