kylecorry31/Trail-Sense #56: Add FDroid link to readme

kylecorry31, 2020-05-14:


dennisguse, 2020-05-14:
This works quite nicely: 

```

<table>
    <tr>
        <th>Free</th>
        <th>Paid (same as free)</th>
        <th>Donations</th>
        <th>Translations</th>
    </tr>
    <tr>
        <td>
            <a href="https://f-droid.org/packages/de.dennisguse.opentracks">
                <img alt="Get it on F-Droid" src="https://fdroid.gitlab.io/artwork/badge/get-it-on.png" height="60" align="middle">
            </a>
        </td>
        <td>
            <a href="https://play.google.com/store/apps/details?id=de.dennisguse.opentracks.playstore">
                <img alt="Get it on Google Play" src="https://play.google.com/intl/en_us/badges/static/images/badges/en_badge_web_generic.png" height="60" align="middle">
            </a>
        </td>
        <td>
            <a href="https://liberapay.com/dennis.guse/donate">
                <img alt="Donate using Liberapay" src="https://liberapay.com/assets/widgets/donate.svg" height="45" align="middle">
            </a>
        </td>
        <td>
            <a href="https://www.transifex.com/opentracks/opentracks/">
                Transifex
            </a>
        </td>
    </tr>
</table>
```

kylecorry31, 2020-05-14:
@dennisguse Thanks for that template!

closed 2020-05-14 by kylecorry31