kylecorry31/Trail-Sense #59: New Feature Ideas

kylecorry31, 2020-05-16:
Please feel free to post new feature ideas in the comments on this issue. I would love to have some other thoughts about what should be included in this app (no guarantee that I will include them, but I will consider each idea). No idea is a bad idea here, worse case I may say that I will not include it in Trail Sense (but who knows, if it is a really good idea, I or someone else may work with you to build it into a separate app). I should also mention, that I am not averse to the idea of having separate companion apps to Trail Sense with specific niches if a good feature does not meet the goal of Trail Sense. 

Some set in stone goals of this app:
- Trail Sense must not use the Internet in any way, as I want the entire app usable when there is no Internet connection
- New features must provide some benefits to people using the app while hiking, in a survival situation, etc. 
- New features should make use of the sensors on a phone rather than relying on stored information such as guides

For reference, some things I have considered in the past and removed were:
- Offline maps 
    - Removed due to Internet use or making users upload map files into the app
    - [This app](https://f-droid.org/en/packages/net.osmand.plus/) already does that very well
- Morse code / strobe / flashlight
    - Removed because it would not be too beneficial for most users (morse code / strobe) and the supported Android versions of this app have a Flashlight built into the System UI
- "Survival blueprints" (recipes for making items while in survival situations)
    - There are a lot of other resources out there and it is best presented in a book format
- Survival guide
    - There are a lot of other resources out there, including this [awesome app](https://f-droid.org/en/packages/org.ligi.survivalmanual/)
- Animal behavior
    - I can't say there is enough science / supporting evidence behind Solunar theory for me to make accurate predictions of animal behavior in general, though if someone provides strong evidence I will consider it
- Tides
    - From what I can tell, the tides vary place to place according to local ocean geography (so not just the moon cycle which is what I initially thought). If this is not the case, let me know - I would also like this feature - though, this is best fit for another app (which can also pull Internet data) since the tide information isn't too important to hikers (open to discussion on this though)

vuori, 2020-05-16:
Saw the app in F-Droid recents and it seems quite useful already. A couple of nice-to-have features that came to mind:

 * Allow as location share target. For example, in OsmAnd when selecting Share > Send, have Trail Sense show up in the app list.
  * Support swipe between tabs.
  * Maybe far-fetched: temperature. I think most phones have at least a battery temperature sensor. The reading will vary based on device activity and location (pocket, desk, etc.), but might produce useful readings with some care in calibration and use.

kylecorry31, 2020-05-16:
@vuori Thank you for the ideas!

1. Yes, the location share target would definitely be a useful feature, I'll add an issue for that. 

2. I'll take a look at this to see if Android supports it with little developer effort. 

3. I actually looked into making a temperature sensor available in my early development (+ humidity), though I found that it was very inaccurate due to the phone heating up from use and being in a pocket / temperature-controlled room. Another thing I noted while looking into this feature though is that temperature/humidity is something that the phone's user would be able to feel without the phone, and given that when looking at it in Trail Sense they would already be able to feel it, I'm not sure how much value that would provide (other than the phone issuing heat alerts/advisories to let the user know that they are at risk for heat exhaustion/stroke or frostbite).

vuori, 2020-05-16:
re: 3. Yeah, I can see the technical issues and put that way giving it a pass makes sense.

kylecorry31, 2020-05-16:
> re: 3. Yeah, I can see the technical issues and put that way giving it a pass makes sense.

I did create a new issue to look into it (https://github.com/kylecorry31/Trail-Sense/issues/62), the heat / cold alert aspect may be a useful reminder to stay cool or warm. I may put it as an off by default setting if anyone wants to try it out though.

swedneck, 2020-05-16:
It might be nice to be able to see in which *direction* the sun and the moon are, perhaps as icons on the compass?

swedneck, 2020-05-16:
Another feature, ripped straight from the compass app on my pebble, is to have the compass change to a "linear" display, like in skyrim, when the phone is held up.  
The current display works okay when held up to your face, but it gets a bit weird when you're facing south and north is shown to be downwards.

kylecorry31, 2020-05-16:
So I purposely made Trail Sense only display in portrait orientation, and didn't design it for landscape. Is having landscape orientation support something anyone would want? If yes, give a thumbs up to this comment.

swedneck, 2020-05-16:
I don't see any harm in supporting landscape (aside from more code), but portrait mode certainly seems the most useful since that's how you'd hold the phone while walking.

Landscape could be very useful for the vertical compass, however.

sethtara, 2020-05-19:
Multiple language support
I can help with that...

AHappyUser, 2020-06-20:
A speedometer and odometer would be a nice addition.

Saijin-Naib, 2020-07-12:
I'd love to see GPS and compass accuracy visualized somehow. Give you an idea if you should be trusting your instrument or yourself more in a given location.

Saijin-Naib, 2020-07-13:
Huge ask:
Geopackage or GPX logging

Saijin-Naib, 2020-07-25:
Ability to take or associate an image with each waypoint.

An optional pop over of the image when the waypoint draws could be nice, as well.

Saijin-Naib, 2020-07-25:
Option to pull data from connected sensors, for instance Fitbit watches.

My phone does not have a barometer, yet my Fitbit does, so having Trail Sense poll my fitbit for atmospheric pressure would be awesome.

https://dev.fitbit.com/build/reference/device-api/barometer/

kylecorry31, 2020-07-26:
@Saijin-Naib Thank you for the ideas! I'll look into them to determine their feasibility and how well they'll fit into Trail Sense and the "no internet policy" of trail sense (in particular the connected devices idea)

Saijin-Naib, 2020-07-26:
Integration as a Share target so locations from text messages, map programs, browser, etc. can be sent to Trail Sense and added as waypoints.

Would make planning trips etc super easy when in a location with data services, though once stored, they obviously no longer need data.

Saijin-Naib, 2020-07-26:
Ability to add notes to a waypoint. Alongside images, this can help navigate in confusing or dense locations, like a multi-path split at a trailhead (picture of spot, and a note to go left around the Spruce, etc.)

Saijin-Naib, 2020-07-26:
Ability to organize waypoints into groups and sub-groups.

Would be great to organize a system of trailheads within a park, for instance.

(Geopackage also could help with this by having tables per group, etc)

Saijin-Naib, 2020-07-26:
Ability to store/show height and angle to waypoint in addition to distance and bearing.

(Geopackage also allows for XYZM points)

kylecorry31, 2020-07-26:
@Saijin-Naib Thanks for the feature ideas! I'll look into them - also, Trail Sense should already be a share target for geo coordinates, is that working for you?

Saijin-Naib, 2020-07-26:
@kylecorry31 , yeah for Geo in specific, I see that it can act as a Share source to things like Magic Earth, which is great.

swedneck, 2020-08-04:
how about a step counter that can be used to calculate roughly how far one has travelled based on your height (which would estimate the stride length)? It could work as a fall-back for GPS positioning, where it estimates your position based on the last known position and how many steps you've taken in what direction.
I imagine this could be very useful for people in dense forests where GPS is flaky.

kylecorry31, 2020-08-04:
@swedneck that is a very interesting idea, though I'm a bit worried about how accurate it would be. Ex. if the user has the phone in their pocket/backpack and is walking, the phone wouldn't be facing in the same direction as the person, hence the estimated location would be wrong (though may be corrected by an incoming GPS signal). I can think of a few other areas which would lead to inaccurate results as well.

I'll do some more research to see how feasible this would be / if I could make an accurate implementation of this - thanks!

swedneck, 2020-08-05:
@kylecorry31 perhaps the user would have to manually set the direction they're walking, and change that direction in the app any time they move another direction? It'd be a bit cumbersome, but i think you'd be pulling up the compass at those points anyways.

whoizit, 2020-08-09:
Hey. That's a cool application. But I don't have the ability to add a lot of points through copy/paste in a certain format. Like that:
```
59.940425 30.313874 Point Description 1
59.923596, 30.385413, Point Description 2
...
``

whoizit, 2020-08-09:
also i want get point from map (OSM in f-droid)

kylecorry31, 2020-08-09:
@whoizit Thank you for the feedback! I do have bulk adding of beacons planned (#95) - I'm not sure when I'll get to it though.

You should currently be able to create a beacon from OSM - you have to share a point using the geo: option and it should give you an option to open Trail Sense. Let me know if that isn't working for you.

whoizit, 2020-08-11:
share point from OSM using the geo: works for me. Thank you very much!

Saijin-Naib, 2020-08-16:
@kylecorry31 this last update is amazing! So many great UX improvements in it.

Thanks so much for your hard work!

kylecorry31, 2020-08-16:
@Saijin-Naib Thank you for the kind words! I'll hopefully be able to spend a bit more time getting the rest of the new features implemented (as seen in the issues tab here). 

So, the accuracy value for the compass comes from Android itself, I'm not sure how that is calculated internally. The GPS, on the other hand, is something that I could display, as Trail Sense receives the GPS accuracy in meters, and I perform a calculation to determine high/med/low (and right now, it only factors in the position accuracy, not elevation accuracy). I'll create a new issue to present this information.

maverick74, 2020-08-25:
About the flashlight: 

I understand that you removed the feature because it already exists on Android but, AFAICT, there is no built-in feature that supports SOS Morse signal, and it would be good to have it available in this app...

kylecorry31, 2020-08-26:
@maverick74 I will consider adding it - not sure how I feel about adding a new tab just for the flashlight, but maybe that would be useful. What do you think?

Another option could be a companion app (I mentioned before that I'm planning on creating a tide / ocean app similar to this), though there are a few nice flashlight apps our there (Simple Flashlight on F-Droid comes to mind)

maverick74, 2020-08-30:
I don't think you need to clutter the ui just for something that "might be useful someday" (let's hope that day never comes!)

I think the important is that the feature is there and that once we get to it (even if we have to take a few steps) we can easily enable/disable it (to save battery, for example)

kylecorry31, 2020-08-30:
@maverick74 After some thought, I decided to implement the flashlight (as you mentioned SOS isn't built into Android - and I shouldn't assume that the flashlight is built in to the quick actions or something) - I actually found a really nice place to put it in the UI (top right corner of the navigation tab), and it actually balanced out the UI with the recent move of the ruler button. I'm going including it in the release tomorrow as well!

For reference, you press the flashlight button once to turn on the light, a second time to turn on SOS, and a third to turn it off - works like a lot of flashlights I've seen. If you turn it on, you can close TS and shut off your screen and the light will stay on (like the native flashlight will do).

Going forward, I'm going to consider what you said that even if we need to take a few steps to enable it, a useful (and potentially life saving) feature should be there when needed. That could mean that I could have an item in settings (for example) which can launch feature X even if you can't access it in other places of the UI (saves on creating tabs for features that don't relate well with the existing tabs or are still being worked on).

kylecorry31, 2020-09-11:
Hi all, I wanted to post this to share that I've started working on the first Trail Sense companion application, TS Maps. 

TS Maps is going to be an offline maps application that closely integrates with Trail Sense so you can easily navigate to your beacons / create beacons from TS Maps. Keeping in line with the goal of TS, the maps app will prompt you to download any terrain maps you want upfront so that it will be completely usable offline (similar to how OsmAnd~ works). 

What it will be:
- An offline maps application targeted towards wilderness treks or survival situations (same as TS)
What it won't be:
- A general maps app for navigating to X location (ex. Google Maps)
- A social maps app (ex. sharing your location/reviews/etc)

So why did I decide to create my own maps application?
- I was not really happy with the user interfaces of existing offline maps applications (free), I kind of want something more Trail Sense themed and easy to use
- I wanted to have complete control over integration with TS - yes, opening a geo: intent with another maps app works fine, but a companion app will have better flexibility here
- I already have most of the core framework in place with Trail Sense, and I've been wanting to extract that into one or more libraries
- I can add whatever features I want (ex. navigate to the nearest water source)
- Can't be a part of TS because it accesses the Internet (and a lot of people have become reliant on TS not doing that, so I'm not willing to add it in TS directly)
- Creating apps is fun :)

Interested in suggesting a feature? Do so here: https://github.com/kylecorry31/Trail-Sense-Maps/issues  It is extremely early in the development process (AKA I just started it a few hours ago), so any ideas are welcome!

Also, I have no idea when I'll be ready for the first production release of TS Maps - hopefully in the next month or two though.

Screenshot after a code dump from TS (fun fact: in early TS development, I had the maps on the navigation tab)
[image]

openStove, 2020-09-15:
I'm often in the montain and I need an accurate altimeter.
I'm missing the ability to manually  tar the altimeter with gps altitude when location is on, or by typing it...

kylecorry31, 2020-09-16:
@openStove I'm not 100% sure what feature you are asking for, I'll list what you can currently do with the altimeter (in v 0.8.0) and see if any of that meets what you are looking for. 

- The altimeter currently uses the GPS to get your altitude, from there it will use the barometer to fine-tune that measurement so as you move it will update the altitude accordingly
- In the altimeter calibration section in the settings, you have the option to enable or disable that barometer fine tuning, as well as switch to above mean sea level elevation (which is different from what the GPS elevation is). You can also override the altimeter by either typing the altitude, using the GPS to get an altitude reading (one time), or entering the current sea level pressure and using the barometer to get the altitude (one time)

I think what you may be asking is to allow setting that baseline altitude (ex. manual override of the altimeter) but then allowing the barometer to adjust that reading as you climb?

If that's the case, I think I can make it so that you can do the following steps:
1. Open the altitude calibration settings
2. Disable auto adjust altitude (which I may rename in this case)
3. Enter your altitude override (typing, from GPS, or from sea level pressure)
4. Enable fine tune using barometer

johnhowe, 2020-09-19:
Hi. Would you consider adding knots (nautical mph) as a distance unit?

kylecorry31, 2020-09-19:
Sure! #208

pegelf, 2020-10-04:
Thank you for that great app. I found it coincidental on the F-droid store and it replaced two apps which I had downloaded on the Play Store before and this one even has more features. I didn't knew that I need moon and sun icons on a compass and now that I have them, I don't want to use other compasses anymore :D.

My feature idea is a bubble level based on the gyro sensor. I've built a kitchen together while only using my phone as a measurement, so it is accurate enough. It can be used completely offline and I think it fits well in an app that uses phone sensors  for displaying useful tools.

kylecorry31, 2020-10-04:
@pegelf Thank you!

So, under the tools tab on Trail Sense, there is an Inclinometer which may do what you are looking for (though it only works for when the device is sideways, not flat). I was actually thinking of doing a more traditional bubble level, almost like what the iPhone has on its compass app (because the compass works best when flat). I can see it being either on the compass itself or on its own page (or even both), as a 2D level may be useful when setting up camp, so you don't end up on a slant at night (sure you could use the inclinometer, but it's not 2D) or if you need to set up a flat surface (ex. for cooking)

pegelf, 2020-10-04:
Oh, I hadn't the experimental features enabled. Thank you for your reply. Now I have found the Inclinometer and it does the same what I would've expected from a bubble level. For me it also doesn't need to be visual bubble level. The display of the degrees is more accurate.

I think the tool could work flat if you would display the X- and Y-inclinations of the phone.

If you want to add an actual bubble level to the app, I think an overlay like the ruler on the compass would look great.

ThracianKnight1907, 2020-10-24:
Hello! Thank you for making this app! I'd like to suggest a feature: have the ability to disable certain notifications. For example, it's probably not necessary to be notified about sunsets when in a city or at home.

kylecorry31, 2020-10-24:
@ThracianKnight1907 Hi, you should be able to disable any of the notifications in the settings of Trail Sense - the one for the sunset is "Sunset Alert", under the astronomy section of settings.

ThracianKnight1907, 2020-10-24:
Indeed, I didn't see it. Found it now

scoson, 2020-10-31:
Hi kylecorry31, I'm just trying out this app for the first time and it's great! Very useful for everything it does without being cluttered or gaudy. I didn't see this mentioned (sorry if already considered) but was wondering if it would be possible for this app to implement a (push to) feature for a telescope? Maybe it could already do that, not sure?? You could connect/strap the phone to the side of the optical tube and while you have an app like celestron skyportal open, get the coordinates  for an object in azimuth/altitude, while having Trail Sense display az/alt in big red numbers. Then you push the telescope where you need to go. Skyportal does display coordinates in compass mode, but very small in upper left corner. I don't know a lot about it, but just throwing it out there, thnx!

jigpu, 2020-11-01:
I was wondering if a feature to "sight" a beacon might be considered. I often find myself being able to see a landmark of interest (e.g. an antenna mast) but not being able to place it on a map. If I were orienteering I could get the bearing to the landmark with my compass and use a rough estimate of its distance to guess the landmark's approximate position. If I wanted to be more precise, I could triangulate the landmark's position by drawing a line from my current position along the bearing, move to a second location, and then draw a second line along the new bearing.

It'd be great if there was some way to do something like this in Trail Sense. I'm imagining the "drop beacon" page having several entry options:

1. Absolute position: enter latitude, longitude, and optionally the elevation (the current beacon entry screen)
2. Bearing position: enter approximate distance and point the phone at the landmark. Beacon lat/lon is calculated from distance and compass reading. You could also integrate the inclinometer to more accurately sight a landmark and calculate an elevation.
3. Triangulate position: point phone at the landmark from multiple locations (again, use of inclinometer as well as compass might be nice). When the first sighting is complete the beacon is added to the "select beacon" screen, but it is initially disabled. You're given the option to add a second sighting from the "select beacon" screen, after which the app can triangulate the position and enable the beacon. The "select beacon" screen might even allow you add even more sightings to refine the triangulated position (e.g. placing the beacon at the centroid of all calculated triangulations).

With these options in place, I could sight a beacon and then open it in something like OsmAnd to see where it is on the map and understand where it is in relation to other features or learn its name (e.g. trying to figure out what mountain I'm looking at).

kylecorry31, 2020-11-01:
@scoson Thank you! I may consider adding that feature (in the tools page), though I don't know how useful it would be for hiking and wilderness survival - if you can find a use case in that area let me know and I will add that feature.

@jigpu Great idea, I'll find the best way to get these features into Trail Sense. Also, something you can do right now is tap the compass on the navigation tab to navigate to a bearing - this doesn't have a distance associated with it though.

Saijin-Naib, 2020-11-02:
@kylecorry31, loving the updates, as well as the Marker Groups!

Because I am ever "that guy", is it possible to allow for nested grouping so that we can make a Group for a Park, then sub-groups for the various trails that are within that park?

Thanks!

scoson, 2020-11-02:
@kylecorry31 I had a feeling it (push to) would be considered too niche anyway. It would have zero use for anything but astronomy anyway but I thought since this app technically has an astronomy section it might be considered. I don't know exactly how many apps out there currently have this function but the folks on the Cloudy Nights forum probably wouldn't mind having another option. Thnx for the great app!

frauhottelmann, 2020-11-17:
It would be nice, if Trail Snse coudl show coordinates in [UTMref / MGRS](https://en.wikipedia.org/wiki/Military_Grid_Reference_System). A lot of government issued maps have this grid system on them. It's very easy to find yourself in such a map if you have the coordinates. The cherry on top would be a way to set the precision level.
Here is a coordinate conversion website: https://coordinates-converter.com/en
Example for Berlin, Germany:  N 52.517037 E 13.38886 = 33U UU 90676 19766
10 digits after the grid square ID is a precision level of 1m which is usually to high for most maps, hence my wish for a precision level selector.

kylecorry31, 2020-11-17:
@frauhottelmann thank you for the idea, I will consider that along with issues #187 and #184

notramo, 2020-11-20:
@kylecorry31 I would like to have a camera view in the compass when the phone is held vertically. This could be used to identify objects for navigation. E.g. I remember the object that is in the direction of North, to navigate using that object (can save battery if it is sufficient to check the phone less often).

notramo, 2020-11-20:
@kylecorry31 I would like to have a feature which uses the camera for bubble level. E.g. show a line at the center of the camera view, and show the direction the phone should be moved to be fully vertical (0 degree). This can be used for metering the altitude of the phone meters away.

notramo, 2020-11-20:
Also, having a high-contrast theme would be useful as it could be used with lower screen brightness (can save battery).

kylecorry31, 2020-11-20:
@notramo I am actually planning on building a Trail Sense AR mode (either a companion app or directly in Trail Sense) (https://github.com/kylecorry31/Trail-Sense/issues/221) 

Other cool applications of camera integration:
- Viewing nearby beacons (with altitude)
- View the path of the sun and moon throughout the day
- Cloud identification (can help determine weather patterns)
- Inclinometer (object height measurement)

I can experiment around with a proof of concept to see how feasible this would be.

I know people are a bit more sensitive about the camera permission, but given that TS doesn't have Internet permissions I don't think it would be a problem requesting that (and on newer android phones, TS can request at runtime so people can deny it if wanted). If added to TS, I may make this disabled in settings by default (as the camera does use more battery / may be overkill for some users).

FTno, 2020-12-01:
I would like to propose a Lightning Distance Calculator - calculate the approximate distance between you and the lightning (measure time between lightning and thunder).
Nice if it could include a history of recent measures.

llllllllqq, 2020-12-08:
Hi,
Could you please consider "GPS Time(date)" feature?
It can get the GPS time and help correct my watch without network.

kylecorry31, 2020-12-08:
@llllllllqq Hi, that feature is already in place for the next release :)  (#306)

llllllllqq, 2020-12-08:
> @llllllllqq Hi, that feature is already in place for the next release :) (#306)

Thank you!
Another feature request,
Is it possible to show a blank board with my track?
It means there's no map, just a grid, and allow users using finger to draw their own map on it (such as the river, the hill). And the show the user's track according to the GPS to keep in mind that which way/area has been explored and mark the dangerous area manually by drawing.
Simply put, it's like a path tracker + drawing board.

[165 more replies]

closed 2023-08-11 by kylecorry31