kylecorry31/Trail-Sense #61: Swipe between tabs

kylecorry31, 2020-05-16:


qwerty287, 2021-04-29:
I implemeted this, I wanted to know how difficult it is to implement touch gestures and because TS has this issue, I tried it in TS. It's working, but the user experience is not the best. I can push my code ti my fork if you want to try it.

kylecorry31, 2021-04-29:
Sure, I would be interested to see how this works - I never implemented it in case I wanted to have a swipe gesture in the future (hasn't happened yet) or change my navigation pattern (already happened)

qwerty287, 2021-04-30:
I pushed it into my TS-Fork, in branch main: https://github.com/qwerty287/Trail-Sense
It has an option in the settings and is disabled, you have to enable it and restart the app. Technically, it just recognizes a touch gesture with a library and compares the values for the x-coordinate - if it's enough, the tab switch is triggered. (The first commit is important, the second just adds the option in the preferences)

qwerty287, 2021-05-05:
Any news on this?

kylecorry31, 2021-05-05:
Sorry, forgot to reply. While it is an interesting way of implementing this that bypasses some of the limitations of the architecture used, I'm not going to merge it into Trail Sense - 1 for the reason listed above (and hence why this issue hasn't been touched) and 2 because of some usability concerns - if this issue were to get done, I'm looking for view pager like behavior, but it gets tricky with some of the deeper navigation with tools. Even if both were resolved I'm still not sure I want it - some of these issues are community suggestions that I recorded and want to have around to consider in the future, but I may choose not to implement them.

qwerty287, 2021-05-05:
Yes, I agree, I wouldn't implement this in my app. But I don't think that there's any other way to get this swipe between tabs - its just not the idea behind the layout with these tabs. (So I don't think this issue is fixable)

kylecorry31, 2021-05-05:
Agreed, I do think the swiping makes sense on the onboarding screen - but users only see that once so it's not a high priority

qwerty287, 2021-05-14:
I'm not sure if it's useful on the onboarding screen.
1. As you said, users see this just once
2. What do you think, a user uses to go to the next page? The big button or the swipe, and there's no note that you can use the swipe? (I don't think any user would use the swipe option, even if there would be a note and a note like "you can use swipes to navigate through these onboarding pages" wouldn't be nice.)

Of course you can implement it - but I don't think anybody would use it.

kylecorry31, 2021-05-14:
No, I don't actually plan on modifying the onboarding screen - I just know that other apps use view pager there

closed 2021-05-31 by kylecorry31