kylecorry31/Trail-Sense #65: Add linear compass display when phone is vertical

kylecorry31, 2020-05-16:
Like Skyrim / Subnautica / battle royale games

kylecorry31, 2020-05-16:
Beacons shown offscreen will need to have an arrow pointing toward them until they are shown on screen

swedneck, 2020-05-16:
It'd be nice to show the direction to beacons on the normal compass as well, should that be a separate issue?

kylecorry31, 2020-05-16:
I had meant the active beacon, which is currently present on the current compass mode. You are referring to displaying all beacons correct?

swedneck, 2020-05-16:
Ah, well maybe the user could select a number of beacons to display?

kylecorry31, 2020-05-17:
Okay, I like that. So the user would be able to show the closest X beacons (unless a beacon is being navigated to, I believe). I don't want to the UI to be too cluttered so I will explore a few different designs for this.

kylecorry31, 2020-05-18:
Here's the progress on this so far:

[image]

kylecorry31, 2020-05-20:
Note to self: customize the linear compass library to match the ticks with the round compass / adjust the UI a bit

closed 2020-05-23 by kylecorry31