kylecorry31/Trail-Sense #69: Barometer diagram: add min/max

dennisguse, 2020-05-19:
At least on the F-Droid version there seems to be no defined min and max for the diagram of the barometric data.
This leads to situations that although there has almost been no change in barometric pressure the diagram looks like there was a big change (in my case 1022 to 1026 only).

Solution: add a lower and upper limit for diagram.

kylecorry31, 2020-05-19:
Agreed, I actually do have that logic in place, just picked too small of a range

kylecorry31, 2020-05-19:
How does something like this look?

[image]

dennisguse, 2020-05-20:
Nice!
According to Wikipedia it usually ranges from 940hPa to 1040hPa on earth. Why not use the full range?
https://en.m.wikipedia.org/wiki/Barometer

kylecorry31, 2020-05-20:
Using the full range makes it difficult to see smaller, yet meaningful changes and may not be fully accurate when not using sea level pressure. I will experiment with that though.

dennisguse, 2020-05-20:
i see :)
However, the weather at home is at the moment rather too good for testing ;)

kylecorry31, 2020-05-20:
I'll put some fake in the UI today and post that later. Looks like there is good weather all week for me too (great for testing the navigation features though!)

kylecorry31, 2020-05-20:
These screenshots are using data from my local area as a storm approached. I personally prefer either the 20hPa or the 40hPa because they clearly show that the weather is worsening (note: the forecasts in Trail Sense are created via changes in barometric pressure in the 3 hour range for short term forecasting and 48 hours for long term forecast trends, so the user should be able to see the chart matching up with the forecast). In my experience, my local weather varies between about 990 and 1030 hPa typically. 

Range: 20hPa
[image]

Range: 40hPa
[image]

Range: 60hPa (normal weather condition variance across earth)
[image]

Range 100hPa (normal atmospheric variance across earth)
[image]

kylecorry31, 2020-05-23:
I'm going to close this - if you feel like the new min and max do not address this properly, feel free to reopen.

closed 2020-05-23 by kylecorry31