kylecorry31/Trail-Sense #71: Weather forecasting open thread

kylecorry31, 2020-05-20:
For those of you using the weather forecasting, how accurate is it for you?

This forecast is only using your phone's barometer, so I would expect some inaccuracies as that is inherent to barometric forecasting, but I can tweak the sensitivity if needed (or put a setting in place if there is mixed feedback).

For reference, the algorithm I am using is:

```
tendency = pressure change over 3 hours
if tendency <= storm threshold, then storm

if pressure <= -fast threshold, then worsening fast
if pressure <= -slow threshold, then worsening slow

if pressure >= fast threshold, then improving fast
if pressure >= slow threshold, then improving slow

else not changing
```

The algorithm was constructed based on various recommendations from many websites. As you can tell, there are a few parameters which can be tuned (storm threshold, fast threshold, slow threshold), though if I were to create a setting for this, I would prefer two settings at max for this: a storm detection sensitivity (already present in latest master branch, though the values are predetermined) and a weather sensitivity value (which will be scale for the fast or slow thresholds).

aw-bib, 2020-05-30:
> For those of you using the weather forecasting, how accurate is it for you?

I did not manage to use it in the app yet, simply cause I'd need to have a gps signal right now _all the time_ which is not possible inside buildings e.g. (I just found a tool that allows me to fake gps incl. the height information.)

However, I do watch this change for quite some time with a "classical" weather station that effectively uses the algorithm outlined. The experience there is that it works quite reliable, as is to be expected (it's a textbook algorithm, right ;), especially, wrt upcoming wind / storm warnings. As soon as the stations alerts go off, I can easily tell that I'll have an official strong wind or gale warning for the area in my mailbox. 

In principle, one could even implement the geostrophic wind scale and, depending on the latitude from GPS, give an estimate for the upper level of the wind to be expected.

The main drawback of my station is (and here your app can improve easily and this rose my interest initially) that one can not read of actual values. It just gives a tendency of relative values for the last 12h and a warning if the levels programmed are reached. Right now I use values from an official site located close by. This requires internet and only works at home as otherwise you'd need to find such a station first etc.pp. Thus, I'd find it nice to have a vertical ruler to your scale, so one could read the time difference and value easily, and probably a simple csv export every _n_ hours.

>  if I were to create a setting for this

I would suggest to go for this as it effectively sets the level (sensitivity) of wind warning (~ pressure gradient). A suitable level can be preset, but it might depend a bit on your surroundings, so it should be adjustable for what you call `fast` / `slow`.

Also for gale warnings a setting for the level would be good. Officially, 10Bft is called a storm, some would argue that an 8 or a solid 6 is quite a bit of wind ;) Plus, the gradient only gives an idea of the geostrophic wind without any reduction due to friction etc. Depending on your surroundings you'll feel much less or almost that value.

> I would prefer two settings

This should be enough given the fact that the gradient is a symmetric thing.

If you do want to improve further you could consider: _do I move towards the centre of the depression or away from it?_ Depending on the direction it either steepens or flattens the gradient experienced. (The effect would be negligible if your speed is << the speed of the low in question.) However, as there is no easy way to tell (on the device, that is) how you're moving it would require manual input for the direction, measurement of the actual speed of movement and, hence probably overcomplicate things.

kylecorry31, 2020-05-30:
Thank you for all that feedback! 

I can definitely add a better vertical scale to the chart (and I can look into adding the time values as well). The CSV export may be nice for some users, I'll put it as a low priority feature request though (technically Trail Sense stores the pressures in CSV format already, it is just in the app's local storage). 

I have to do more research into the geostrophic wind scale to see if it would be feasible to derive from a phone's sensors, but it would definitely be useful if it could be done. 

As for the heading into the storm vs away that could be possible (given that the user has the appropriate settings enabled for GPS). I can do more research into that, but it will be low priority for now.

aw-bib, 2020-07-03:
It would probably be a nice addition if the weather _notification_ would not just tell that the app is monitoring in the background, but also give some data like current pressure, tendency over the last 3h. Maybe reusing the forecast icon from the main screen and it's title:

- 🌤 1015 ↗ (+1 hPa/3h)
- 🌩 990↡ (-7 hPa/3h)

Also, I'd suggest that tapping the notification guides you to the weather page instead of the compass, but this may depend on the general idea for the notifications.

closed 2020-08-02 by kylecorry31