kylecorry31/Trail-Sense #72: Add translations

kylecorry31, 2020-05-20:


pr0gr8mm3r, 2020-05-27:
Sine version 0.3.0 the app's name has been translated (for German at least) by @paulle in Pull request #83 or somewhere prior. This made me wonder if fhe app name should even be translated, as the English name sounds way more modern to me. However, I can also understand if people want it to be in their language, I just think there should be guidance added on this topic in the Contributing section of the Readme so it can be done in the same way for all languages.

paulle, 2020-05-27:
I believe the app name shouldn't be translated. It's OK with an app name in English. Depending of language the app name can become much larger.

-------- Original-Nachricht --------

closed 2021-01-10 by kylecorry31