kylecorry31/Trail-Sense #74: Testing Trail Sense Debug Builds

kylecorry31, 2020-05-20:
You can get debug builds for Trail Sense here: https://www.dropbox.com/sh/gplqslml3iyq2ec/AACn0Arv3KRHAUF1wCusmoZ2a?dl=0

I'll post a new comment here each time I build a debug build so you can download it and provide feedback on some new features while they are still in development. 

(I wasn't sure where to post the debug builds, so I chose Dropbox because it was convenient - let me know if there is a preferred free software platform I can use though!)

sethtara, 2020-05-20:
[image]

### Little UI changes required

sethtara, 2020-05-20:
[image]

#### having  ruler at side is nice but having it by default at first glance doesn't look that great..

kylecorry31, 2020-05-20:
@sethtara Thanks! I totally agree with that, I was thinking of adding a button to slide out the ruler and putting a background on it so it doesn't mess with the linear compass. In the meantime, I added a setting to show/hide the ruler.

sethtara, 2020-05-20:
[image]

#### How about adding the language option a start..

sethtara, 2020-05-20:
[image]

### some UI tweaks needed

### I tried to fix it 
[image]

kylecorry31, 2020-05-20:
A new version has been uploaded containing a version of the astronomy fix

kylecorry31, 2020-05-21:
A new version has been uploaded containing translations and the ability to turn off weather monitoring in the background via settings.

kylecorry31, 2020-05-23:
A new version has been uploaded. Contains setting (off by default) to display multiple beacons (WIP, see that issue for more details) - early feedback welcome!

It also contains a ruler scaling factor setting that may be needed for some devices.

kylecorry31, 2020-05-25:
#### New debug build
- Beacons can be marked as visible
- Setting to determine how many beacons are visible

kylecorry31, 2020-05-30:
#### New debug build
- New astronomy graph (fixes inaccuracies with sun/moon positions)
- Support on astronomy page for places where sun/moon doesn't rise or set or does so out of order
- Set and rise times on astronomy page are ordered by occurrence instead of always being in the same order
- New setting to display the sun and moon on the compass

kylecorry31, 2020-06-06:
#### New debug build
- Added compass mode setting to put more options there in case the legacy or experimental didn't work

kylecorry31, 2020-08-02:
**New debug build**

- Added ability to put notes on new beacons (still need to add the ability to edit existing beacons and needs a little UI work - you can click the 'View Notes' text when navigating to a beacon)
- Improved the sea level pressure algorithm (needs some testing)
- Did some internal database changes - would be useful to see if anyone runs into any errors (looks good on my end)

kylecorry31, 2020-08-02:
**New debug build**

- Improve the UI of the accuracy indicators (and adjusted navigation UI accordingly)
- Bug fix for older app versions upgrading to the new database

kylecorry31, 2020-08-03:
**New debug build**
- Improve navigation UI
- Add estimated time to beacons
- Add accuracy tips (click on accuracy icons)
- Add setting to adjust weather forecast sensitivity
- Fix bug when trying to update the visibility of beacons

kylecorry31, 2020-08-15:
**New debug build**
- Add weather information to the notification
- Add ability to edit beacons (long press from the beacon list)

kylecorry31, 2020-08-24:
**New debug build**
- Added ability to hide the weather notification (in settings)
- Adjusted button placement on the navigation page

kylecorry31, 2020-08-27:
**New debug build**
- Added flashlight to navigation tab (press once for light, a second time for SOS, and a third time for off) (runs in background as well)
- Fixed lag when opening TS

kylecorry31, 2020-08-31:
**New debug build**
- Added ability to manually override location, altitude, and declination as well as tweak some more sensor values
- Improved speed calculation (ignoring inaccurate values)
- Added compass calibration dialog
- Persist selected beacon on close / tab change
- Added setting for displaying the pressure and tendency
- Added tendency to the weather page
- Added about section to settings

kylecorry31, 2020-09-07:
**New debug build**
- Added setting to force pressure updates
- Adjusted tidal range calculation
- Added experimental features (inclinometer, tidal range)
- Potentially fixed a bug causing crashes on some Samsung devices

kylecorry31, 2020-09-23:
**New debug build**
- Barometric altimeter
- Temperature and humidity
- More accurate astronomy data
- Show dusk/dawn times
- Pressure setpoint (tap and hold the current pressure on the weather tab)
- Show beacons within a certain radius
- Allow custom weather update interval
- Better Android 11 support
- Bug fixes

kylecorry31, 2020-09-25:
**New debug build**
- Potential fixes for null pointer exceptions on some phones
- More translations

kylecorry31, 2020-10-03:
**New debug build**
- Added beacon groups
- Changed sea level pressure algorithm - should be more accurate now (some advanced settings for this are still WIP)
- Reorganized settings
- Added animations when switching between views (not the tabs yet)
- Changed how readings are obtained in the background (adjustable with the weather settings - to test the new changes, you probably want to have "force weather updates" off and "update weather in foreground" on)
- Data storage changes
- Improved back button behavior

I'm looking for some people to test this version for a few days to verify the following features work better than the previous version:
- There should be less missed weather updates
- The sea level pressure conversion should be better - you can adjust the settings for this in the "Barometer and thermometer" settings

This version is not backward compatible with the previous version, therefore I made a new "Beta" build which will install alongside whatever version of Trail Sense you currently have, when you are done testing you can uninstall it and the real Trail Sense will be untouched. (The icon of the beta build is blue and has beta in the app name)

kylecorry31, 2021-09-15:
There's a new beta version for 2.9.0

The following services are using a new background scheduling process and I would like feedback on if they are behaving the same. Please keep an eye out for if the GPS gets stuck on, I believe I fixed it and haven't experienced it in several days, but if it does happen I would really appreciate a bugreport.zip file. 
- Weather monitor (> 15 minutes)
- Backtrack (> 15 minutes)

Other new features:
- Lunar eclipses
- Improved solar panel alignment algorithm
- Astronomy details are clickable
- Supermoon (when you click an astronomy detail for the moon or long press the datepicker and search for supermoon)
- Weather data can be exported to CSV, which is useful for bug reports on the weather feature (weather settings) - the export does not contain sensitive data (pressures, altitudes, temperatures, humidity, and time)

The debug APK will overwrite your existing Trail Sense installation, and the beta APK will install as a separate application.

Thank you!

kylecorry31, 2021-10-09:
There's a new beta version for 2.10.0

It contains the following changes:
- Log button on Weather tab to log a reading without running the background service
- Improved cloud tool
- Cloud scanner (experimental) - determines the type of cloud (leave feedback here: https://github.com/kylecorry31/Trail-Sense/issues/968)
- Material theme update
- Humidity history (for devices with a humidity sensor) - accessed through the thermometer tool
- A background compatibility mode (experimental) for devices having trouble with the background services (weather monitor, backtrack)

I only uploaded the beta APK since there were database changes meaning you couldn't revert to an older version of Trail Sense - the beta app will install as a separate app and will not affect any data you already have in TS (and can be uninstalled with no problem)

There are a few minor UI tweaks remaining before the final build which I plan to release in a day or two.

FlatChest, 2021-10-09:
I've installed it with background compatibility mode enabled. I'll edit this post in a few hours to see if the weather monitor works now. Thanks again!

qwerty287, 2021-11-20:
@kylecorry31 You could try to automate this process by using GH actions - currently you build the app, but do not use the output. You could improve this by changing the app's packages and similar things so that a beta/debug build is generated that can be installed along with the stable version, then build the app (what it's currently doing), sign the APK and upload the APK to Dropbox or something else (found a Dropbox uploader: https://github.com/marketplace/actions/dropbox-uploader).

You would have to find a way to store the keystore for signing, but GH actions should provide something like this. 

This would:
* allow simple testing without much effort
* use GH actions for something instead of just building the app and doing nothing with the APK
* allow testing of the latest commit's build

This would be kind of a win-win-situation: you don't have to build the app yourself + you can test the builds from the latest commits

kylecorry31, 2021-11-20:
@qwerty287 that's a good idea, I'll play around with it. The only downside is that when I create a build for testing, I normally have specific things that I want to be tested, and can point out areas that are still in progress / broken so people know what to test or not, as to not waste their time.

ACABMAN666, 2023-08-17:
> You can get debug builds for Trail Sense here: https://www.dropbox.com/sh/gplqslml3iyq2ec/AACn0Arv3KRHAUF1wCusmoZ2a?dl=0
> 
> I'll post a new comment here each time I build a debug build so you can download it and provide feedback on some new features while they are still in development.
> 
> If you are interested in getting a debug build, please leave a comment on this issue and I will create a new version and say what's in it. I stopped generating them automatically due to lack of feedback/engagement.

Hi, I'd be interested in testing the new features with the debug build

kylecorry31, 2023-08-17:
@ACABMAN666 thank you!

I'm going to replace these debug builds with #1922 

As of right now, I don't have any new features in need of testing (nothing major this release) - but I will likely have something in the next release (full export), so I'll plan to complete #1922 in a couple weeks.

closed 2023-08-17 by kylecorry31

ACABMAN666, 2023-08-17:
> @ACABMAN666 thank you!
> 
> I'm going to replace these debug builds with #1922
> 
> As of right now, I don't have any new features in need of testing (nothing major this release) - but I will likely have something in the next release (full export), so I'll plan to complete #1922 in a couple weeks.

Ok thanks for the information

kylecorry31, 2023-08-25:
Hi all - nightly builds of Trail Sense are now available, see https://github.com/kylecorry31/Trail-Sense/discussions/1940 for more details.