kylecorry31/Trail-Sense #80: Make monitoring weather in background optional

operat, 2020-05-21:
Monitoring weather changes in background can be useful in certain situations, but should be optional, because it contantly sucks power from your battery. Most times you don't need that featuer anyway, as you just want to check the compass or sun times etc. It's tedious to force the app to stop everytime you had checked the compass, in order to "switch off" the monitoring. Focus should be on saving battery power, especially with outdoor activities in mind. Thus, everything that uses power constantly, should be optional.

kylecorry31, 2020-05-21:
Hi,

The ability to stop this was added in a recent version (#54) - it may take a few more days for that to reach F-Droid (I submitted the update 6 days ago). The way that works is it will start each time you open Trail Sense and you can stop it via the notification (and restart it by opening Trail Sense). I will also add a setting to opt-out of this in general. 

As per the battery used by this feature, I can only speak to the Pixel 3A, but according to the battery monitor it uses about 1 - 2% per day to have continuous weather monitoring (it takes a new reading every 10 minutes - I may make that customizable).

kylecorry31, 2020-05-21:
There we go:
[image]

I'll be creating a new release in the next couple of days, in the meantime you are welcome to install the debug builds of Trail Sense to get this feature earlier.

closed 2020-05-21 by kylecorry31

kylecorry31, 2020-05-23:
@operat FYI, the latest version in F-Droid now contains the ability to stop the background monitoring via the notification (the always off setting hasn't been released just yet)