kylecorry31/Trail-Sense #85: Support landscape orientation

kylecorry31, 2020-05-23:
Would make sense for linear compass

aw-bib, 2020-08-26:
Also the barograph on the weather page could profit from it.

kylecorry31, 2021-04-23:
Not going to support landscape orientation - too much to port over.

kylecorry31, 2025-01-23:
Looks like Google will be forcing this on larger screens. There's 2 years before fully required but will take a lot of effort to support.

Trail Sense does not have official tablet support, so I need to test what is within the actual supported limits of TS.

https://android-developers.googleblog.com/2025/01/orientation-and-resizability-changes-in-android-16.html

kylecorry31, 2026-11-02:
Closing until Google expands requirements, I'm not interested in supporting landscape beyond what I do for large devices. I don't use Trail Sense like that personally (and this is my hobby project) and that just makes all new features take more time to ensure they work on small screen landscape.

closed 2026-11-02 by kylecorry31