kylecorry31/Trail-Sense #90: Option to show only beacons within a certain distance

swedneck, 2020-05-25:
Not sure how this would combine with other visibility settings

kylecorry31, 2020-05-25:
I was thinking of adding this, I'll have to explore some ideas here

kylecorry31, 2020-08-31:
For now just to get this in place, I may need to do the following:
1. User clicks on max beacon distance setting
2. Switches to a new screen where they can enter the distance + select the unit that they are providing the distance in (respects distance unit settings)
3. Clicks done, brings them back to the setting list

I know that breaks the user experience of the settings page, but it is difficult to deal with different units in the settings screen - I'll eventually revisit and create a custom dialog where they can set all this stuff.

commit c7eab70e (2020-09-23): Add visible beacon radius

Closes #90
closed 2020-09-23 by kylecorry31