kylecorry31/Trail-Sense #91: Better display for when sun doesn't set/rise

swedneck, 2020-05-25:
Currently if you live somewhere the sun doesn't rise/set for a full day the astronomy page simply shows the rise/set times as 00:00/23:59

It should instead make it obvious that the sun simply doesn't rise/set that day.

kylecorry31, 2020-05-25:
Definitely, I recognize those as the default values I'm returning in those cases - I can make it more clear (like the moonrise/set dash to indicate it doesn't happen)

kylecorry31, 2020-05-25:
Still have some more work to do here (like when sun rises or sets, but not both - and fixing that position issue)

[image]

closed 2020-05-27 by kylecorry31