kylecorry31/Trail-Sense #93: Sun position appears to be wildly innacurate

swedneck, 2020-05-25:
When I have the sunset/rise setting as 'actual time' and 'civil twilight' the sun is almost at the bottom of the circle, despite clearly being near the horizon.

I presume this is because it's not actually showing the sun's position in relation to the horizon, but it's still kind of confusing.

kylecorry31, 2020-05-25:
Base sun and moon 'clock' on the altitude of the sun and moon (with the peak altitude being at solar noon) rather than the percent between rise and set.

kylecorry31, 2020-05-25:
This will be tricky when the sun doesn't rise or set. Thinking out loud here so I don't lose these ideas when developing this. 

If the half-circle (line between black and blue areas) represents the horizon, then the sun will not cross it in that case - I need to do more research into how the sun moves in areas where it doesn't set or rise. My assumption, which may be wrong, is it goes from low in the sky to high in the sky each "day", and that should somehow be reflected in the UI.

Google's weather app makes it a lot easier to show changes like this (they have a line graph type display, which looks like a sine wave). I do like the circular UI that I have here, but if I can't find a way to accurately represent this case, then I may have to switch to something like that. The first thing I can experiment with though is dynamically moving the horizon line on the circle so that the bottom of the circle is the minimum altitude of the sun during the day, and the top is the maximum, and the color change would occur at the altitude of 0. That will probably break the moon's inner circle though, and it will need to have its own coloring. 

The more I write here, the more I'm starting to think that the sine graph is a better representation of the sun and moon position (and accuracy is more important than aesthetics in this case. I'll create a simple UI mockup of that sometime this week.

kylecorry31, 2020-05-25:
Something like this - it shows from 12AM to 11:59PM (0:00 - 23:59) on a given day - which is actually what the circular clock in Trail Sense currently shows - and like you said, it is confusing and unclear that it does that. 

[image]

kylecorry31, 2020-05-26:
Yeah, this feels much better:

[image]

swedneck, 2020-05-26:
That looks great!

kylecorry31, 2020-05-27:
To do: Handle overlap of sun and moon on chart (maybe move moon up or down and insert a line pointing to where it belongs on the chart, as the sun is more important to display on the line)

closed 2020-05-27 by kylecorry31