kylecorry31/Trail-Sense #103: Tides

aw-bib, 2020-06-01:
> From what I can tell, the tides vary place to place according to local ocean geography (so not just the moon cycle which is what I initially thought).

As you already noticed this _is_ complex and you can't really get anything that useful by just the moon cycle. Indeed, tides seem to have a touch of an arcane art to it. As far as I get it the main problem is that the hydrographic offices (except the US and they cover only the US) don't publish their harmonics data and/or you need to licence them.

However, it would definitely be nice to have them on board. Some random ideas:

- The most widely used free application that centres around tides is probably [XTide](https://flaterco.com/xtide/) which seems to be used entirely / in parts / with some extensions and changes in quite a few applications. One could consider porting it and make sure one can easily pass on the necessary harmonics-files. However, this might be quite a bit of work in porting and unearthing the mentioned harmonics files for non-US areas is a bit of an adventure.
- The only really free prediction tool I found for Android is [TidePredictor](https://arachnoid.com/android/TidePredictor/index.html). It is GPL, so sources are available and it could be added to your project. This tool gives, to my experience, quite reliable values. The only drawback I found in TidePredictor is, that it could use some better (mobile friendly) user interface.
- The simplest idea to get really good data on board might be hooking up with "the authorities". Some of them offer tide data for a week or so from their web interfaces. As this may be _good enough_, one could imagine some function that just fetches the HTML and stores it for later display employing some bookmark mechanism where one could probably hit "update" before heading out. This does require internet, but only on a minimalistic level. AFAIK UKHOs EasyTide also offers an API to fetch data and it does offer ports all over the world. (At least there seems to be a plugin for OpenCPN along these lines in current development.) Some examples of such services for further research:
    - [UKHO EasyTide](http://www.ukho.gov.uk/easytide/EasyTide/SelectPort.aspx): worldwide
    - [SHOM, France](https://maree.shom.fr/): worldwide
    - [NTSLF, UK](https://www.ntslf.org/tides/predictions): UK/Ireland
    - [NOAA Tidal predictions, USA](https://tidesandcurrents.noaa.gov/tide_predictions.html): USA
    - [BSH, Germany](https://www.bsh.de/DE/DATEN/Gezeiten/gezeiten_node.html): Germany
    - [Rijkswaterstaat, Netherlands](https://www.rijkswaterstaat.nl/kaarten/astronomisch-getij.aspx) Netherlands

isRelatedTo: #59

kylecorry31, 2020-06-02:
Thank you, I can look into this a bit further, but I may not put this functionality into Trail Sense. I am working (slowly) on a companion app for diving/ocean sports, it may be best to put it there: https://github.com/kylecorry31/Free-Diver (name TBD)

kylecorry31, 2020-06-28:
I'm going to close this issue here, but I am interested in building it out in another app related to ocean stuff.

closed 2020-06-28 by kylecorry31