kylecorry31/Trail-Sense #104: German Translation

paulle, 2020-06-02:
There are two things in the I can't find in the German string for translate. 
The ruler calibration and the waxing (for the moon).
[image]
[image]

kylecorry31, 2020-06-04:
Hello, I added the German translation (using Google Translate) of the ruler calibration: https://github.com/kylecorry31/Trail-Sense/commit/7161a25d8132917b9fd80450b9d150033513b29a

Does that look correct?

Also, I didn't translate the moon phases though, would you be able to help with that?

The phases in English are:
- Waxing crescent
- Waxing gibbous
- Full moon
- New moon
- Waning crescent
- Waning gibbous
- First quarter
- Third quarter

paulle, 2020-06-04:
Will try it.
Full moon  Vollmond
New moon  Neumond
First quarter  Erstes Viertel
Third quarter  Drittes Viertel
Waxing gibbous  zunehmender Dreiviertelmond (means: Waxing gibbous moon) in German there is no expression for only Waxing gibbous.
Waxing crescent  zunehmender Sichelmond
Waning crescent  abnehmender Sichelmond
Waning gibbous  abnehmender Dreiviertelmond 
waning  abnehmend
waxing  zunehmend

Hope this will help you

commit 2f9d2999 (2020-06-06): Add German translations for moon phases

Closes #104
closed 2020-06-06 by kylecorry31