kylecorry31/Trail-Sense #106: publish APK along with releases

swedneck, 2020-06-02:
would this be possible? I haven't gotten any updates from F-Droid so it'd be nice to be able to just grab the APK from the github release page

kylecorry31, 2020-06-02:
Yeah, F-Droid takes a while - I can definitely post the APK there!

kylecorry31, 2020-06-02:
Done - I posted an unsigned release build from Android Studio (should be about half the size of the normal debug builds.

closed 2020-06-02 by kylecorry31

swedneck, 2020-06-02:
Hmm, it seems i'm unable to install the release APK.
Sadly i don't get any sort of error, it simply says "app not installed"... Thanks android!

kylecorry31, 2020-06-02:
That's odd - maybe it has something to do with a version conflict with the currently installed version. I'll post the debug APK as well, maybe that will help

swedneck, 2020-06-02:
i thought so too, but even uninstalling the previous version i had doesn't make the newest release work

kylecorry31, 2020-06-02:
That is odd, let me know if the debug version works

swedneck, 2020-06-02:
Yeah debug version works, very odd