kylecorry31/Trail-Sense #114: Linear compass changes compass direction when using experimental compass

MSBack, 2020-06-04:
When I tilt the phone up, so the linear compass shows, the compass direction changes up to 180 degrees, all the way from north to south. When I hold the phone horizontally again, the compass direction changes back immediately.

This issue only occurs when using the experimental compass.

kylecorry31, 2020-06-04:
Hello, does this occur when the linear compass setting is off?

MSBack, 2020-06-05:
Yes, it seems to occur regardless of whether the linear compass setting is on. Also, it seems that this behavior is seen with both the "regular" compass and the experimental compass, but it is more pronounced in the latter case.

I don't know if it's relevant, but I am using a Nokia 3 phone running Android 9.

MSBack, 2020-06-05:
This stackoverflow thread seems to have some relevant information: https://stackoverflow.com/questions/16317599/android-compass-that-can-compensate-for-tilt-and-pitch

And this issue from another app with compass functionality: https://github.com/cgeo/cgeo/issues/5889

kylecorry31, 2020-06-05:
Thanks, I will look into those options and see if I get any improvements. 

I'm not seeing this issue on my end, so it may be something specific to your phone. When I have some updates here, would you be able to test out some debug builds to help me resolve it?

MSBack, 2020-06-06:
Great! Yes, just let me know.

kylecorry31, 2020-06-06:
@MSBack I just posted a new debug build here https://github.com/kylecorry31/Trail-Sense/issues/74

I added a new setting called "Compass Mode". From there, you can select the "Custom" value and try that out (that was from the stack overflow post).

Let me know if that works.

kylecorry31, 2020-06-06:
FYI, I am seeing much smoother transitions when moving from flat to vertical (and also the code is much cleaner with the "Custom" compass)

MSBack, 2020-06-06:
I have installed the new debug version, but I will test it tomorrow. Thank you for this!

kylecorry31, 2020-06-09:
Hi, just wanted to see if that fix resolved this issue?

kylecorry31, 2020-06-13:
I'm going to assume that this fixes the issue, as I am noticing much more consistent behavior. If it does not work for you, feel free to open this issue again.

closed 2020-06-13 by kylecorry31