kylecorry31/Trail-Sense #121: showing velocity

paulle, 2020-06-15:
Showing the velocity in an extra tab would be a nice option. Showing The actual velocity and the mean of velocity in last hour (or a graph showing the velocity of a certain time period for example 1 or 2h or the last 30minutes).

kylecorry31, 2020-06-16:
That would be interesting and easily obtainable (as I'm already getting GPS updates in the background).

What would this be used for though?

paulle, 2020-06-16:
It can be an interesting information when you are walking/hiking in hard or not so hard areas under various weather conditions, tired or not to see your real velocity. It can help you to see how long you may need to arrive at a certain waypoint and you can better estimate the influence of area/weather conditions.

kylecorry31, 2020-06-20:
For now, I will consider adding the average velocity over the last X hours as well as the distance traveled to the compass screen. In the future, I may add a new tab, as you mentioned. 

(Though, work has been pretty busy lately, probably won't get around to much Trail Sense work for a couple of weeks)

kylecorry31, 2020-06-20:
https://github.com/kylecorry31/Trail-Sense/issues/59#issuecomment-647005249

paulle, 2020-06-20:
Thank you for this, and the nice app in general.

I imagine it's a lot of work coding an app. I don't understand nothing of coding so can only translate a little bit or report bug.

-------- Original-Nachricht --------

kylecorry31, 2020-08-03:
This could also be used to give an approximate ETA when navigating to beacons.

Ex. 
~1h
away

kylecorry31, 2020-08-03:
I can currently show the velocity of the phone (not average) with little effort, but when adding average velocity/distance traveled it will either need to have a new service created (meaning new notification / background task) or it can piggyback off of the monitor weather service (which the user can disable). More work on that will be needed - I'm going to go with the simple solution for now.

paulle, 2020-08-03:
Showing only the actual velocity is an advantage, even more when it can be done with a simple solution.

kylecorry31, 2020-08-04:
[image]

The time estimate to the beacon adjusts for the potential of not traveling in a straight line by slightly increasing the distance and it also uses the average velocity of the user.

paulle, 2020-08-04:
> [Screenshot_1596543473](https://user-images.githubusercontent.com/5938560/89292905-13106200-d62b-11ea-96b9-6ba801f8fefd.png)

You are very fast and I like your screenshot a lot. What do you think about making the font size for altitude and speed bigger and bolder so it will be easier to get these values at a quick glance. I could imagine this could be an advantage while walking or riding a bike.

Thank for your work.

-------- Original-Nachricht --------

closed 2020-08-15 by kylecorry31