kylecorry31/Trail-Sense #125: Show when sun and moon reach peak position

kylecorry31, 2020-07-05:


paulle, 2020-07-21:
This would be a good feature.

kylecorry31, 2020-08-20:
I will add a 'more' button to the astronomy page where some of the following info can be found (or just make the details list scrollable - have to see what looks better):
- Solar noon
- Solar midnight (not sure how important)
- Lunar noon
- Lunar midnight (not sure how important, or what it is called - got to do some research here)
- Moon illuminance (remove from phase name in regular UI)
- Tidal information (whether it is a neap, spring, or average tide - not sure how important)
- Sun/moon altitude (degrees)

All values will respect the selected day, except when on the current day where they will reflect the value at the exact time.

closed 2020-08-20 by kylecorry31