kylecorry31/Trail-Sense #154: Hight / Velocity / Accuracy readings

aw-bib, 2020-08-26:
I get a bit confused with the readings.

Right now, Trail-Sense tells me:

- Hight:
  - `-4m` by GPS and this sometimes even changes even though I do not move. Plus: right now it does not have any GPS signal, so the +/-1  or so can not result form GPS errors I guess.
  - It tells me 153m by barometer which is equally confusing. However, to get a height by barometric pressure one would need some sort of calibration in the first place and I understand this is still on the todo list.
  - None of those values should actually reflect my real height at the moment. Unfortunately, I have no real value but the average for the town I live is +6m, while the terrain map of the area gives something around 30m.
  - Use `FakeGPS` to set the readings to some position, manual hight: `30m` and I get a reading of -10m. Again a bit of a variation. I would expect the `30m`.
- Barometer: the tendencies are perfect, just the readings are a bit off, and the curve is very nice and helpful.
 - If I set _sea level pressure_ to `off`, given that I almost live at sea level, i get something in the right ballpark. Right now it reads 995 hPa, the official reading of a nearby meteorological station corrected for sea level is `997.2` so this sounds reasonable.
 - Set _sea level pressure_ to `on`, it jumps to 1117hPa, which sounds unreasonable.
- Velocity: as mentioned I do not move, it still gives some old reading 1.3km/h slightly changing. Maybe there should be some sort of _reset_ button.

kylecorry31, 2020-08-26:
Hi,

Hopefully this helps:
- Height / altitude
    - When set to GPS in settings, it gets the initial reading from the GPS, then updates using the barometric changes using the GPS altitude as the baseline
    - Unfortunately, the barometer in itself is bad at determining altitude unless the sea level pressure is around standard atmosphere - the calibration will help with that
    - The GPS provides the altitude of the user above a model of the Earth that doesn't factor in local variations - I have a table of corrections which are applied to the raw GPS elevation to get the true elevation. I got the corrections from another open source application, and did not verify them myself. Based on the fake GPS reading, it looks like your correction factor was -40m, which would make the true reported elevation equal to 36m in your first point. I'm going to add a setting to enable/disable this altitude correction - I'll create the issue for it.
- Barometer / sea level pressure
    - The readings are most likely off due to the altitude reported in your first section
- Velocity
    - The velocity shown is reported directly by the GPS (not the average used for ETAs), and in areas where the GPS signal may be poor the GPS drift may be reported as moving. I can look into this to see if there's anything I can do to improve that.
- Accuracy
    - That is odd, both the accuracy and calibration (figure 8) are handled by Android itself (no additional logic in Trail Sense). I'm going to do more research here (unfortunately, I can't reproduce it) - and maybe this will require a setting to invert the compass (wonder if some devices have their magnetometer readings flipped or something). Do you have the legacy compass on in settings?

kylecorry31, 2020-08-26:
Closing this in favor of #159, #158 and #157.

I'll run the numbers on the sea level pressure readings to ensure that their cause was the altitude correction - if not, I'll create another bug for that. 

Thanks!