kylecorry31/Trail-Sense #155: Forecast / storm sensitivity

aw-bib, 2020-08-26:
Right now one can set the sensitivity to low/medium/high, but one does not have any idea what gradient is actually used in the back end. I would suggest to allow some numerical input here, like it's done at my usual weather station. Something like `nn hPa/3h` or `mm hPa/4h`. I do understand that this might require a bit more knowledge at the users end.

kylecorry31, 2020-08-26:
I'll consider making it numerical, but for now, I can display the x hPa/Xh in the summary of the preference (below the title)

aw-bib, 2020-08-27:
Inspirational(?) notes on numbers from the manual of my other hardware (just as I finally happen to have it at hand, and it works quite well for some years now):

> - The change of weather forecast icon is in accord to the relationship between current relative pressure and the pressure change since the last `12 hours`. If the weather is changing, [some indicator]. And after the next `3 hours`  if weather conditions become stable [other indicator].

(They use 4 symbols _sunny_, _partly cloudy_, _cloudy_, _rainy_ and animated arrows or just arrows between them to indicate the change from to. This is very similar to Trail Sense :)

> - The pressure threshold can be set to suit the user's requirement for weather forecasting from `2-4 hPa` (default `3 hPa`). Areas that experience frequent changes in air pressure require a higher setting compared to an area where the air pressure is stagnant. [...] e.g. if `4 hPa` is selected there must be a fall or rise in air pressure of at lest 4 hPa before the weather station will register this as a change in weather.

=> I would guess this is low/medium/high in the forecast sensitivity.

> - The storm threshold can be set to suit the users requirement for storm forecasting from `5-9 hPa` (default 6 hPa). When there isa fall over pressure threshold _within 3 hours, the storm forecasting will be activated [...] for 3 hours.

(I found there seem to be two schools of thought, one relying on 3h intervals others on 4h.)

commit 9d34a991 (2020-09-01): Add pressure tendency amount to sensitivity settings

Closes #155
closed 2020-09-01 by kylecorry31