kylecorry31/Trail-Sense #156: Suggestion: add some actual readings

aw-bib, 2020-08-26:
The improved notification in 0.6 and the feature to jump to the barograph is really nice. :)

I do notice that sometimes the icons used in the notification area and on the Weather page differ. E.g. I get a sun as notification icon (improving very soon) vs. a clound (improving soon) on the Weather page.

Also I would suggest to supplement the text "Improving very soon" by some actual reading like `996 hPa (+4 hPa/4h)`.

isRelatedTo: #124

kylecorry31, 2020-08-26:
Thanks, I also noticed the icons were different - that's because on the weather page it uses the current pressure to determine if that is high/low for the given altitude to get more accurate icons. I'll port that functionality over to the notification as well. I also like displaying the actual reading, will do!

aw-bib, 2020-08-27:
Also a tendency for some hours is a very valuable thing to have, especially if it might get a bit windy. ;) I way already about to suggest to have the gradient for _n_ hours (n=3, 4 or some setting) also displayed along the ruler on the weather page.

commit 134f654f (2020-08-31): Add pressure and tendency to notification

Closes #156
closed 2020-08-31 by kylecorry31