kylecorry31/Trail-Sense #158: Ignore drifting velocity

kylecorry31, 2020-08-26:
GPS drift may cause an incorrect velocity to be shown - validate that the user is moving before displaying a velocity (ex. either with multiple velocity readings, GPS position changes, or accelerometer).

aw-bib, 2020-08-27:
If velocity is derived from GPS one would probably need to handle the case that GPS is off:

- one could derive the velocity from the pedometer (envisioned in suggestion #142)
- add some brackets/color/symbol that makes clear this is a reading that is ancient/is not updated because of ...
- switch it off

Maybe I'm the only one switching off GPS all the time to conserve battery ;)

kylecorry31, 2020-08-27:
I agree that something should be done if the GPS is off - I think for this issue, I'm going to just hide it from the UI. If I choose to implement the pedometer, then I'll revisit it.

kylecorry31, 2020-08-31:
This may already be fixed in the latest release - I improved the GPS accuracy by waiting for a specific number of satellites to be achieved or a timeout to have been reached before reporting a valid reading. That should also improve speed accuracy. 

I can add one more check here to see if the GPS has a reported speed accuracy, and if it does and the variation is greater than the current speed then it will ignore that speed (I'll have to do some research with this).

commit 503a5612 (2020-08-31): Factor in speed accuracy

Closes #158
closed 2020-08-31 by kylecorry31