kylecorry31/Trail-Sense #159: Compass points south on some devices

kylecorry31, 2020-08-26:
See #154 

Possibly create a setting to invert the compass / research why this could happen.

kylecorry31, 2020-08-27:
If this is an issue with the compass on the device - maybe it would be best to calculate the accuracy myself as Android was returning high accuracy.

kylecorry31, 2020-08-27:
https://stackoverflow.com/questions/58055288/android-sensor-accuracy

aw-bib, 2020-08-27:
> Possibly create a setting to invert the compass / research why this could happen.

The `North points South` was just an example for _way off_. IOW not just a few degrees off or _I missed out on the variation_ kind of thing. It's a random inaccuracy that can be cured easily by the usual _figures of 8_. Also please note: this is a common feature through other apps as well. [SatStat](https://www.f-droid.org/en/packages/com.vonglasow.michael.satstat/) e.g. shows the same behaviour as does [OsmAnd](https://f-droid.org/en/packages/net.osmand.plus/) e.g. (A reason why I use the latter usually "north up".)

Just to avoid a miss-understanding; it's not that the compass is reversed systematically.

If I didn't use trail sense (or any other android compass) for some time I will just have to recalibrate it before it gives any sensible reading. (Unlike its the trusty iron companion. ;) In #154 I was just wondering how it could be that the accuracy reading tells me _high_ or _medium_ in this case. I found this a bit confusing. From the stackoverflow article you just linked it became a bit clearer.

To handle it an idea could be that there is some output/warning/hint that says: _your last calibration is <time> ago, I don't really know where north is, please some figures of 8 to make sure._ It would make sense anyway as some metal might have appeared in the vicinity that was not there an hour ago.

kylecorry31, 2020-08-27:
Thanks! I like the idea of offering a calibration notice (which is something that actually should be done every time the compass is used on a smartphone). I'm relying on Android's calibration to do this, not sure if I can tell when the last calibration was - maybe something I could do is display a dialog which prompts the user to calibrate after they select a beacon to navigate to (with a way to disable it in settings / from within the dialog) or have them pop up if the accuracy is unknown/low/med and it hasn't been displayed in X time.

I'm also going to look into doing the accuracy calculation myself if that is possible.

aw-bib, 2020-08-27:
> calibration [...] something that actually should be done every time the compass is used on a smartphone

I have to admit that it took me quite a while to fiddle out how to actually do it at all ;)

> I'm relying on Android's calibration to do this, not sure if I can tell when the last calibration was 

If the OS doesn't offer it one could think of some timestamp by some fake calibration button or call up of the compass / a beacon etc.

> display a dialog which prompts the user to calibrate after they select a beacon to navigate to (with a way to disable it in settings / from within the dialog) or have them pop up if the accuracy is unknown/low/med and it hasn't been displayed in X time.

Sounds also like an interesting idea. Something like a hint with a "don't show me again" checkbox.

thoelze1, 2021-01-18:
New user here! I like the idea of notifying the user somehow that regular calibration is necessary: I just used the compass for the first time and, if I had been disoriented or lost, I would have trusted the erroneous reading without a second thought.

> If the OS doesn't offer [calibration accuracy]...

I too was hoping that Android might provide calibration confidence somehow, but a note in the [docs](https://developer.android.com/guide/topics/sensors/sensors_position#java) reads

> Note: When testing your app, you can improve the sensor's accuracy by waving the device in a figure-8 pattern.

which suggests the burden of calibration is on the user. This does make me wonder how e.g. Google Maps reports its calibration confidence. Anyways, I like the idea of a hint-style notification to the user as a simple way of communicating that calibration is necessary.

Thanks for your work! I look forward to using the app :)

kylecorry31, 2021-01-24:
@thoelze1 Thank you for the feedback! (sorry I missed your comment earlier)

I have #175 which should bring more attention to this - but again that only uses the manufacturer's reported accuracy (either low, medium, high, or unknown). I'm assuming Google Maps derives its own accuracy (and I've seen a patent that document's how Apple does this to produce a degree value).

Pandapip1, 2023-06-30:
I have a Samsung S10+. After about a day or two, the compass permanently has its accuracy listed as "Poor" despite having seemingly no problems. I suspect that it's an OS issue, but I'll add my info here just in case.

kylecorry31, 2023-06-30:
@Pandapip1 just to confirm, that's a day or two after calibrating it that this happens?

Also, the next time this happens, can you check the Diagnostics tool (Trail Sense Settings > Diagnostics) to see if Trail Sense is able to provide more details? I'm mainly looking for a poor magnetometer, accelerometer/gravity, or gyroscope being reported. 

For reference, on my Pixel 5A the gravity sensor reports as poor but since the magnetometer is good, the compass reports as good (expected). I'm only aware that Android has built in magnetometer calibration (the figure-8 movement), and the other sensors don't seem to support system wide calibration.

Given there is seemingly no issues with that, I'm assuming Samsung ages the accuracy based on the last figure-8 calibration.

Pandapip1, 2023-06-30:
That's really weird. When I initially opened the diagnostics, it was showing the compass as poor. However, about a quarter of a minute after I enabled the background location permission and activity permissions (in case that fixed it), the warning disappeared and the compass accuracy started indicating good accuracy again. I don't know if this is due to opening the diagnostic tools or changing the permissions.

kylecorry31, 2023-06-30:
Hm, that is odd - my guess is it is just a coincidence, that by having it open and using the compass the phone triggered an auto calibration. 

Neither of those permissions are related to the compass and the diagnostic tool doesn't do anything special.

kylecorry31, 2024-12-27:
Not going to implement this.

Reasons:
- Not technically feasible / will be inaccurate
- Frequency of use (ex. edge case) vs effort to implement (Android already reports this, though accuracy may not always be correct, upstream bug)

closed 2024-12-27 by kylecorry31