kylecorry31/Trail-Sense #160: Missing translation German (not in the string)

paulle, 2020-08-26:
This is a screenshot from the astronomy tab.

[image]
It should be    bis Sonnenuntergang   not  bis sonnenuntergang

These are some screenshots from the configuration tab.
[image]
Monitor weather      Wettermonitor
in the background     im Hintergrund
Forecast Sensitivity  Vorhersagegenauigkeit

[image]
Keep sun/moon centered     Sonne/Mond zentriert lassen
When up                              Wenn aufgegangen

[image]
When up                              Wenn aufgegangen
Never                                  Nie
Always                                Immer

kylecorry31, 2020-08-26:
Thanks! It's hard keeping up with translations, especially because I can only really verify English and a tiny bit of Spanish - always afraid Google Translate will translate to something bad/completely wrong.

closed 2020-08-26 by kylecorry31

paulle, 2020-08-27:
Yes, Google translate produces from time to time very strange translations. For me it won't matter to translate/correct the string when a new version with additional text is out, but I don't know how to get access to text not included in the string.

-------- Original-Nachricht --------