kylecorry31/Trail-Sense #164: New translations needed for v0.7.0

kylecorry31, 2020-08-29:
The following strings were recently added for the 0.7.0 release (which will likely be tomorrow evening ~August 30 at 20:00 EDT). Any translations that come in after that will be considered for a 0.7.X release or included in the 0.8.0 release. The bolded words below do not need to be translated, they are there to give context and the %s is a placeholder for a value so leave that in the translation. Feel free to just comment the translations below, and I can handle putting them into the source code. I'm trying out this new format for receiving translations for TS, hopefully, it works well. Thanks!

**Sunset alert**
30 minutes
1 hour
1 hour 30 minutes
2 hours
2 hours 30 hours
3 hours
Sun is setting soon
The sun will set at %s
Sunset alerts
Sunset alert
Sunset alert time

**Coordinate format**
Coordinate format
Decimal Degrees
Degrees Decimal Minutes
Degrees Minutes Seconds

**Flashlight onboarding**
And easily access your flashlight

**Astronomy details**
Solar noon
Lunar noon
Moon phase
Moon illumination
Sun altitude
Moon altitude

**Tides**
Tidal range
Neap
Spring
Normal

aw-bib, 2020-08-30:
Please find here the German translations in front of the english texts.

**Sunset alert**

`30 Minuten` - 30 minutes
`1 Stunde` - 1 hour
`1 Stunde 30 Minuten` - 1 hour 30 minutes
`2 Stunden` - 2 hours
`2 Stunden 30 Minuten` - 2 hours 30 hours
`3 Stunden` - 3 hours
`Die Sonne geht bald unter` - Sun is setting soon
`Sonnenuntergang um %s` - The sun will set at %s
`Sonnenuntergangsalarme` - Sunset alerts
`Sonnenuntergangsalarm` - Sunset alert
`Sonnenuntergang: Alarmzeit` - Sunset alert time

**Coordinate format**

`Koordinatenformat` - Coordinate format
`Dezimalgrad` - Decimal Degrees
`Grad und Dezimalminuten` - Degrees Decimal Minutes
`Grad, Minuten, Sekunden` - Degrees Minutes Seconds

**Flashlight onboarding**

~ `Und einfacher Zugriff auf die Taschenlampe` - And easily access your flashlight

(Sounds a bit strange due to the `And`, might depend on context.)

**Astronomy details**

`Sonnenkulmination` - Solar noon
`Mondkulmination` - Lunar noon
`Mondphase` - Moon phase
~ `Mond: beleuchter Anteil` - Moon illumination
`Sonnenhöhe` - Sun altitude
`Mondhöhe` - Moon altitude

Notes: 
- I understand that `noon` here refers to the maximal height of ...
- I used a different construction in German for `Moon illumination`. One could translate more literally as `Beleuchteter Anteil des Mondes`

**Gezeiten - Tides**

`Alter der Gezeit` - Tidal range
`Nippzeit` - Neap
`Springzeit` Spring
`Mittzeit` - Normal

Given the other terms, I think _Tidal range_ refers the [range variation of the tide](https://en.wikipedia.org/wiki/Tide#Range_variation:_springs_and_neaps) (Neap/Spring etc.) then my translation is correct.  If you refer however, to the hight difference between high and low water in meters, feet, fathoms etc. (what I know as tidal range, but this could be my limited English) the German word would be `Tidenhub`.

HTH.

kylecorry31, 2020-08-30:
Thank you, your assumptions were correct - I should have been more clear when I posted the translations needed as to their context (maybe screenshots could help?)

kylecorry31, 2020-08-30:
For reference, the flashlight onboarding string appears on the navigation onboarding as such: (at top) Place beacons and navigate back to them   (at bottom) And easily access your flashlight

aw-bib, 2020-08-31:
With your additional comments you may use all the above German text strings, even those marked with ~ are fine.

paulle, 2020-09-01:
Here seems to be a little error:
2 Stunden 30 Minuten - 2 hours 30 hours
see above in sunset alert.

aw-bib, 2020-09-01:
> Here seems to be a little error:

Note the error is in the original English part not in the translation.

kylecorry31, 2020-09-01:
Thanks! A notification 32 hours before sunset wouldn't be too useful :)

paulle, 2020-09-01:
I noted it. But better to correct an error before it shows up in the final product.

-------- Original-Nachricht --------

closed 2020-09-24 by kylecorry31