kylecorry31/Trail-Sense #174: Explore companion apps

kylecorry31, 2020-08-31:
It would be cool to have companion apps which TS can link to and open:

1. Ocean (or Nautical) Sense - basically an ocean version of Trail Sense which can tell tides and depth of phone (if in a waterproof case) - very WIP / not really started yet (except for the depth mode)
2. Offline maps - may be able to start navigation to a beacon from TS and open in maps, or view TS beacons
3. TS DEM - an elevation addon to TS, may take up a lot of space hence not going to be included in TS directly

kylecorry31, 2020-09-21:
https://developer.android.com/training/secure-file-sharing/setup-sharing

khimaros, 2020-11-22:
I'm very interested in tide information. How practical is that to calculate offline?

kylecorry31, 2020-11-22:
@khimaros hi, please see #203 for the reasoning why tides won't be included - long story short, tides are really complicated and almost certainly require internet to be remotely accurate

aw-bib, 2021-01-02:
- [Markor](https://github.com/gsantner/markor) actually makes a pretty good companion app to Trail Sense for note taking, inventory, todo and stuff. Especially, as for note taking at some point one would probably want to _get them out_. Similarly, for an inventory / packing list one might want _to get it in_ using your favourite flavour of `vi`. This makes an approach to store plain files on some (private cloud) storage quite charming. Given Markor uses Markdown one also has some formatting available and a pretty decent handling for online links. (Weather forecasts might never be available offline.) isRelatedTo: #351 
- [Tiny Weather Forecast Germany](https://codeberg.org/Starfish/TinyWeatherForecastGermany) might be worth consideration if it comes to fetch an actual weather _forecast_. Despite it's name, it draws data from ICON which is the [_global model_](https://www.dwd.de/EN/weather/weather_world/weather_world_node.html) of the official German meteorological institution. (People outside of Europe might have to cross check it in their area, I do not know how dense the stations are there.) It might make a good companion as it is especially conservative with online data (calculations update only every 6h, so no need to update quicker...), does not require you to be online to display anything, draws in a pretty detailed report for a week in one go and can draw in several forecast spots if you want to. ICON is usually pretty reliable for at least 3 days.
- [Tide Predictor](https://arachnoid.com/android/TidePredictor/index.html) could do some offline tide prediction. Comparing it to the official tables for some time now, it gives a pretty good idea and was at least the best free tool I could find. Times are usually off a bit (up to 30min or so, usually +/-10min), but tidal prediction in my area is known to be notoriously difficult. As for height predictions I did not find a note about the reference point used. In my area it reports almost consistently off by -0.5m compared to the official tables. So I _think_ it is based on _mean spring water low_ (the chart datum used before ~2005), which in my area is just +0.5m above LAT (the chart datum used in current tables in Europe). So, depending on requirements it can well be _good enough_. It uses the harmonics of XTide, and as the harmonics for the US are published it should be almost spot on there. (To be checked by locals.) isRelatedTo: #103, #185, #203

closed 2021-01-10 by kylecorry31

Lee-Carre, 2022-01-01:
See also [#353 (comment)](http://github.com/kylecorry31/Trail-Sense/discussions/353#discussioncomment-1774444) in which I mentioned many (mostly related to surveying).