kylecorry31/Trail-Sense #177: Allow custom weather update interval

kylecorry31, 2020-09-01:
Longer intervals will reduce the accuracy of the forecast but also decrease battery consumption.

The current interval is set to ~15 minutes

Options:
- 15 minutes
- 30 minutes
- 45 minutes
- 1 hour

Beyond that, it is hard to get enough data to calculate the pressure tendency accurately.

kylecorry31, 2020-09-07:
There was, it switched to using system alarms to do it's processing for better battery life as they don't have to fire exactly every 15 minutes. When you look at the graph, are you seeing intervals greater than say 20 minutes?

kylecorry31, 2020-09-07:
If the case is that there were missed updates, I have an idea. I can add a setting which forces an update every X minutes instead of letting the system move them around a little.

aw-bib, 2020-09-07:
> There was, it switched to using system alarms to do it's processing for better battery life

Ah, ic.

> When you look at the graph, are you seeing intervals greater than say 20 minutes?

I understand that clicking on the graph jumps to the measured values. I see the following updates: 0m ago, 2h ago, 3h, 5h, 14h, 16h, 17h, 19h. etc. (taking the current graph which has advanced some steps).

kylecorry31, 2020-09-07:
Okay, yeah it appears there were missed updates, which explains the jagged graph. Interesting, the previous version (0.6.0) had missed updates on my phone, and this version is a lot better. I'll put a force update in 0.8.0, that should fix this issue.

commit 91b190ff (2020-09-12): Allow custom weather update interval

Closes #177
closed 2020-09-12 by kylecorry31