kylecorry31/Trail-Sense #185: Tidal range display

aw-bib, 2020-09-04:
0.7 introduced the display of the tidal range neap/spring/normal which is very nice :) 

Checking the values however, I was wondering how spring/neap is done. Usually, I would expect spring around full/new moon, neaps around first and last quarter with a period of ~4 days for each phase (`normal` is sometimes shorted to 3 days due to solar vs. lunar days).

In Trail Sense I see e.g. `normal` for 3rd and 4th of September 2020, while _the official tables_ give a spring from 2nd  (full moon) till the 5th, then a `normal` from 6th..9th and neap from 10th (3rd quarter)..13th so I think the spring/neap period in Trail Sense is shorter?

Note that I use tables for the German Bight. In this areas we see a delay of tides of some days as the North Sea is a bit special compared to the free oceans. See e.g. [Tide timing](https://en.wikipedia.org/wiki/Tide#Timing)

isRelatedTo: #125

kylecorry31, 2020-09-04:
Thank you, maybe I was a bit too ambitious with this feature - I didn't consider how local variations could affect this as well. The algorithm is really simple: is the moon is full/new then it is a spring tide and same for neap tides and the quarter moons.

The algorithm should really be if the moon was full in the last ~80-90 hours, then spring. I'll try to address this and compare against some tide tables.

With that said, I may mark this feature as experimental under the newly added setting in 0.8.0

closed 2020-09-06 by kylecorry31

aw-bib, 2021-04-01:
> is the moon is full/new then it is a spring tide and same for neap tides and the quarter moons.

BTW: this isn't too bad, especially if you allow for a spring delay of _n_ days. One would need to look up the _n_ required for the current place, but they are actually tabulated or you can derive it from an old fashioned printed or electronic tide table by checking the lowest low / highest high next to full/new moon.