kylecorry31/Trail-Sense #189: Indicate when GPS is acquiring lock

kylecorry31, 2020-09-07:


aw-bib, 2020-09-07:
One could also think of hooking up with tools like [SatStat](https://www.f-droid.org/en/packages/com.vonglasow.michael.satstat/) to load off the display of satellites etc. similar to the function in OsmAnd.

paulle, 2020-09-07:
There is another fine app on f-droid which provides a lot of GNSS information if needed or wanted.
https://f-droid.org/de/packages/com.android.gpstest.osmdroid/
A simple hint popping up within Trail Sense, that a lock is acquired would be fine.

kylecorry31, 2020-09-07:
Yeah, I think for now I'll just put a pop up which says the lock was acquired (and make the accuracy reading say 'Aquiring lock' or something like that)

commit 769f3711 (2020-09-10): Add gps lock message

Closes #189
closed 2020-09-10 by kylecorry31