kylecorry31/Trail-Sense #190: Show twilight times on astronomy tab

kylecorry31, 2020-09-07:


aw-bib, 2020-09-07:
Given the tables get lengthy on said page, maybe it could be integrated with the graph? E.g. giving it a ruler like the weather page that is magnetic and jumps to the next event like noon, rise, set... and show thet time.

kylecorry31, 2020-09-12:
I'm wondering if I should make the altitude and azimuth values available when you click the sun/moon rather than always displaying them there - for the most part they aren't that useful (and you can see the sun and moon on the compass already, so the azimuth is a bit redundant).

That would make more room for putting the twilight times in that list - as the list is more visible than tapping the chart to see the values.

closed 2020-09-12 by kylecorry31