kylecorry31/Trail-Sense #194: Improve sea level pressure when changing altitudes

kylecorry31, 2020-09-07:
Just hiked a mountain, and the sea level pressure readings were off (again - thought I fixed this previously). 

I want to spend a decent amount of time refining this algorithm because:

1. Using the recorded GPS altitude adds noise to the graph
2. Sometimes the GPS returns the same altitude as the last reading (even when it changes)
3. The barometric pressure can indicate changes in altitude (fast changes in pressure)

kylecorry31, 2020-09-07:
If anyone has any algorithm suggestions, I would love to hear them!

Current algorithm:
1. Obtain pressure and altitude history
2. Set altitude to first altitude reading
3. For each pressure, if it jumps drastically it is an altitude change that caused it
4. If there is an altitude change, update the altitude using the change in pressure
5. If new altitude is close to the GPS altitude, use the GPS altitude
6. Convert pressure to sea level with the derived altitude

aw-bib, 2020-09-22:
- v0.8
- Settings / Altimeter
- _Set override from barometer_
- key in _the official reading_(tm) for sea level pressure given by an official weather station. Say this reads `1010.4 hPa`
- Trail Sense calculates an altitude override, say `23.8 m` = 24m - this sounds sensible.
- Compass tab shows `24m` - seems to have arrived.
- Weather shows `1014 hPa` => I would expect `1010 hPa` as I keyed it into the correction, even if things go through the recalculation by altitude etc.pp.

I wonder if this is related.

kylecorry31, 2020-09-22:
Yes, the pressure on that tab is based on altitude history (in my development version, I changed it to display the current pressure regardless of history at the top, and the graph uses the history), so this is related

kylecorry31, 2020-09-28:
I created a new algorithm, and so far it seems to be working better:
- Group readings by altitude and time (essentially it creates groups where the altitude hasn't changed in a while) (configurable by a maximum change threshold)
- If there aren't enough readings in a group, consider it as traveling (configurable by a dwell parameter)
- For each non-traveling group, set the altitude to the average altitude
- For each traveling group, set the altitude of each reading to linearly scale between the closest non-traveling groups
- Convert the pressures to sea level with the derived altitudes

kylecorry31, 2020-10-02:
I also added a setting to ignore traveling readings, so you won't see large spikes in the graph (or get storm alerts) while changing altitude (for example, climbing a mountain generates storm alerts). If you are at the same altitude for around an hour (going to be configurable), then you'll start seeing pressure updates again.

kylecorry31, 2020-10-08:
Remaining work:
- Add settings for dwell duration, max altitude change, and max pressure change

commit 2013b89c (2020-10-10): Add settings for sea level adjustment

Closes #194
closed 2020-10-10 by kylecorry31