kylecorry31/Trail-Sense #198: New translations v0.8.0

kylecorry31, 2020-09-09:
New translations have been added in v0.8.0 - see [strings.xml](https://github.com/kylecorry31/Trail-Sense/blob/master/app/src/main/res/values/strings.xml) for the English strings that were added - feel free to create a PR or post translations here (sorry for not pasting them directly into this issue - there were a lot of things that were added/changed and the best tool for that is the Android Studio translations editor)

I'm going to create the new release tomorrow evening (America/New_York timezone)

aw-bib, 2020-09-22:
I do not yet get how this is supposed to work. Say, I translate the XML. If I just do this and create a PR you'd loose the original text. So there has to be a language flag or the like.

kylecorry31, 2020-09-22:
Yeah, sorry - I still need to find a good way to get community translations in. Essentially there is a folder for each language and that's where the updated files go, so if they are just sent to me I can put them in where they belong.

aw-bib, 2020-09-24:
> Essentially there is a folder for each language

Probably prepopulate that from the English XML and giving a pointer might be easy enough, if people feel comfortable editing the XML (which AFAIK can be done right on github.)

(Note that I'd be fine with a PR as well as some other tools you mentioned.)

This issue seems superseeded by #213

kylecorry31, 2020-09-24:
Closing this in favor of #213

closed 2020-09-24 by kylecorry31