kylecorry31/Trail-Sense #203: Request: Add high and low tide times to astronomy graph.

jacob-i-skinner, 2020-09-11:


kylecorry31, 2020-09-11:
Hi, the only thing I could potentially do around this is put approximate tide times. 

I mentioned in the new features issue that tides are most likely beyond the scope of Trail Sense, and I may create a new app for them at one point.

kylecorry31, 2020-09-12:
I did give this a second look - I experimented with what would happen if I gave the user the ability to calibrate tides to their location. This would essentially involve the user entering the tide times for today, and it would use them to add a time offset for when the moon rises/sets. In my testing, this was only accurate for a few days before the tide times started differing by several hours rather than minutes. 

I think that without harmonics data this feature can't be implemented.

I wonder if it would be possible for the user to enter harmonics data: https://tidesandcurrents.noaa.gov/about_harmonic_constituents.html 

Here's a good book on the topic - I may take a closer look at this: https://tidesandcurrents.noaa.gov/publications/Tidal_Analysis_and_Predictions.pdf

jacob-i-skinner, 2020-09-12:
I appreciate you looking into this so thoroughly. Given the typical use cases it does seems like spinning off tide info into its own app is a good idea.

closed 2020-09-12 by jacob-i-skinner