kylecorry31/Trail-Sense #208: Add nautical distance units

kylecorry31, 2020-09-19:


qwerty287, 2021-04-18:
Isn't this already imemented?

kylecorry31, 2021-04-18:
It isn't - this was to create a feet / nautical miles selector in the settings, I have an implemented because my current architecture really only supports metric and imperial, and I don't really think this would be useful for hiking - with that said, I think this issue can be closed and the current distant converter unit will do.

closed 2021-04-18 by kylecorry31

kylecorry31, 2021-04-18:
@qwerty287 I'm going through the issues in Trail Sense now, and will add a help-wanted label to any issue that I'm comfortable with someone creating a PR for at any time given the details in the issue (if you are interested in working on others, feel free to leave a comment on it and we can discuss/I can share more details) - I'll also try to update the descriptions/comments to make the stories more clear. That should make it a bit easier to figure out which I still want implemented / have a clear idea of what I want.

I appreciate all the development work you've been doing, thanks for the help!