kylecorry31/Trail-Sense #212: Weather update interval

aw-bib, 2020-09-22:
A change in the update procedure for the barograph from v0.6 -> v0.7 resulted in basically missing updates to the graph resulting in a quite jaggered graph with unequal spacings (cf. https://github.com/kylecorry31/Trail-Sense/issues/177#issuecomment-688106894).

v0.8 seems to have changed the update procedure and given the issue reported I also set _Improve weather accuracy_. I understand, that this should force an update "every 20min or so". Running since yesterday, however, I stil see updates like 5min ago, 16min ago, -3h ago, -12h ago, -14h ago. Note that the step 3h->12h was just the night. From observation it seems that a new reading is added to the graph only if I fire up the barograph page manually and if the time difference to the last update is not too long.

Trail Sense 0.8.0 on Sony XPERIA. Trail Sense is even set as an exception from the power saving features, no Stamina mode or the like, just _Adaptive battery_ is on. (I understand that Sony devices seem to be quite aggressive there.)

Note that there were no changes in those settings between v0.6 and v0.8 and I got a smooth graph on the old version. So smooth that I was about to suggest to actually reduce the updates a bit. ;)

isRelatedTo: #177

kylecorry31, 2020-09-22:
Okay, I'll add back in the older version using the foreground service so you can swap between them - that will force the notification to be on.

kylecorry31, 2020-09-23:
Added this back in as a setting

closed 2020-09-23 by kylecorry31

aw-bib, 2020-09-24:
It took some time to track it down on my end, especially to make sure it is working now. Sorry for this delay.

Luckily I have here two almost identical devices one Sony XPERIA  XZ1 (G8341) and one XZ1 Compact (G8441). They are set up identically, same software and updated to the most recent Android available for both of them. Even the apps installed are the same so I can exclude some interference there. Both run Trail Sense, the XZ1 may be one release behind the Compact. 

i noticed the jagged graph first on the compact (my own) one with v0.7 and as mentioned above it persisted in v0.8. Now, I got my hands on the big brother again to install the necessary updates and was about to update TrailSense v0.8 as well. I assumed it on v0.6 so it did not make me wonder that the barograph was ok. _But_ on this one ran v0.7 already, and: no staircase, but just a smooth graph as expected and as you noticed on your phones as well.

I went through all the related settings and discovered one difference: as noted above on my phone with the jagged graph I added Trail Sense _as an exception_ to the power management. I have to do this with one or the other app that does jobs in the background, most noticible DAVx5, that even issues a warning on a Sony. If I don't add it it will not sync. So I instinctively added Trail Sense as it should monitor in the background. However, I missed out to set it on the big brother.

And indeed: _removing_ Trail Sense from the exception list of the power management and all seemed fine. After some hours, then I cleaned all data from Trail Sense and let it run entirely in the background for the last two days to make sure it is fine now and the issue is _fixed_, indeed.

So it seems to be more an entry in the FAQ to check this kind of settings. (I'm pretty ignorant on the market so I don't know how other makes handle the stuff.) On my XPERIA it's just install Trail Sense, _don't think_ and the logging it will work as expected. I don't even need to set the _improve accuracy_, it will just log as expected.

Sorry for being a bit late, I was just offline for the last two days.

kylecorry31, 2020-09-24:
Thank you for the very detailed update. I'll be sure to document that in the FAQ. 

I added the foreground service back just in case, which is enabled through a setting if needed (on my phone the foreground service doesn't work as well)