kylecorry31/Trail-Sense #213: Translations needed

kylecorry31, 2020-09-23:
**Dialogs**
- Copied to clipboard

**Tabs**
- Tools
    - German, Portugese (Brazil)

**Disclaimer**
- Disclaimer
    - German, Portugese (Brazil)
- Trail Sense is a tool that can be used in a survival situation, but you should not rely solely upon it. Please validate it's accuracy, avoid dangerous situations when possible, stay calm, and use common sense.
    - Portugese (Brazil)

**Accuracy**
- Low
    - German, Portugese (Brazil)
- Medium
    - German, Portugese (Brazil)
- High
    - German, Portugese (Brazil)
- Unknown
    - German, Portugese (Brazil)
- Tips for improving accuracy
    - Portugese (Brazil)
- GPS: Go outside with an unobstructed view of the sky. 
    - Portugese (Brazil)
- %s of actual location   
    - Portugese (Brazil)
- %s of actual altitude 
    - Portugese (Brazil)
- %s satellites
    - Portugese (Brazil)
- Compass: Rotate device in a figure 8 pattern and move away from metal.
    - Portugese (Brazil)
- GPS lock acquired

**Flashlight**
- Flashlight
- Tap to turn off
- SOS

**Destination panel**
- ~ %s away
    - Portugese (Brazil)

**Beacon list**
- Edit beacon
    - German, Portugese (Brazil)
-Long press a beacon to view options.
    - Portugese (Brazil)
- Send
    - German, Portugese (Brazil)
- Copy
    - German, Portugese (Brazil)
- Copy beacon
    - German, Portugese (Brazil)
- Copy coordinates
    - German, Portugese (Brazil)
- Maps
    - German

**Create beacons**
- Invalid name
    - German, Portugese (Brazil)
- Invalid latitude
    - Portugese (Brazil)
- Invalid longitude
    - Portugese (Brazil)
- Invalid elevation
    - Portugese (Brazil)
- Elevation feet (Optional)
    - Portugese (Brazil)
- Elevation meters (Optional)
    - Portugese (Brazil)
- Notes (Optional)
    - Portugese (Brazil)

**Weather**
- %.0f °F
- %.0f °C
- %.0f%% humidity
- No humidity data
- Dew point %s
- Monitoring weather in the background
    - German, Portugese (Brazil)
- Storm Alert
    - German, Portugese (Brazil)
- A storm might be approaching
    - German, Portugese (Brazil)
- Storm alerts
    - German, Portugese (Brazil)
- Alerts
    - German, Portugese (Brazil)
- %s %s - %s ago
    - Portugese (Brazil)
- Set: %s - %s ago
- %s / 3h
    - Portugese (Brazil)
- in
    - As in the unit inches
    - German, Portugese (Brazil)
- Heat danger
- Heat exhaustion/stroke risk. Be sure to stay hydrated, stay indoors or in shade, limit exercise, and watch for signs of heat exhaustion or stroke.
- Hypothermia/frostbite risk. Be sure to stay hydrated, dress warm and limit exposed skin, and watch for signs of hypothermia or frostbite.
- Heat alert
- Heat warning
- Heat caution
- No temperature warning
- Hypothermia caution
- Hypothermia warning
- Hypothermia danger
- Temperature accuracy is affected by your phone's heat.

kylecorry31, 2020-09-23:
Feel free to comment as many translations as you like and I can add them to the app. The stuff in bold doesn't need translations, and the sub-bullets indicate which languages already have translations.

The %s or %.0f like strings are placeholders for values.

kylecorry31, 2020-09-23:
(and there are more translations needed than just this, I'll add them later - it takes a very long time to copy them over - and yes I need to find a better way of doing this)

kylecorry31, 2020-09-23:
If I were to create a Transifex (https://www.transifex.com/) project for Trail Sense would people use it to provide translations? Looks like it may be free for open source projects (but I haven't really looked into it much)

yes = thumbs up this post, no = don't respond or thumbs down

raphaelventura, 2020-09-24:
Hi @kylecorry31 I'm willing to help and consolidate the french translation of Trail Sense by doing a whole review : as of version 0.8.0, some strings don't have any translation, and there are some inconsistencies between some translated strings.

A very nice collaborative tool I'v been using on another project is [weblate](https://weblate.org/en/), you can check it out and see if it fits.

paulle, 2020-09-24:
here translation into German

Dialogs

    Copied to clipboard  In die Zwischenablage kopiert

Tabs

    Tools  Werkzeuge
        German, Portugese (Brazil)

Disclaimer

    Disclaimer
        German, Portugese (Brazil)
    Trail Sense is a tool that can be used in a survival situation, but you should not rely solely upon it. Please validate it's accuracy, avoid dangerous situations when possible, stay calm, and use common sense.
    Trail Sense ist ein Werkzeug, welches in Überlebenssituationen verwndet werden kann, aber Sie sollten nicht nur einzig auf es verlassen. Bitte validieren Sie seine Genauigkeit, vermeiden Sie jede gefährliche Situation wenn möglich und bleiben Sie ruhig und vernünftig.
        Portugese (Brazil)

Accuracy

    Low  Niedrig
        German, Portugese (Brazil)
    Medium  Mittel
        German, Portugese (Brazil)
    High  Hoch
        German, Portugese (Brazil)
    Unknown  Unbekannt
        German, Portugese (Brazil)
    Tips for improving accuracy
        Portugese (Brazil)
    GPS: Go outside with an unobstructed view of the sky.
        Portugese (Brazil)
    %s of actual location
        Portugese (Brazil)
    %s of actual altitude
        Portugese (Brazil)
    %s satellites
        Portugese (Brazil)
    Compass: Rotate device in a figure 8 pattern and move away from metal.
        Portugese (Brazil)
    GPS lock acquired  GPS lock erforderlich

Flashlight

    Flashlight  Taschenlampe
    Tap to turn off  Antippen um auszuschalten
    SOS  SOS

Destination panel

    ~ %s away
        Portugese (Brazil)

Beacon list

    Edit beacon  Wegmarke editieren
        German, Portugese (Brazil)
        -Long press a beacon to view options. Langer Druck auf eine Wegmarke um Optionen zu sehen.
        Portugese (Brazil)
    Send  Senden
        German, Portugese (Brazil)
    Copy  Kopieren
        German, Portugese (Brazil)
    Copy beacon  Wegmarke kopieren
        German, Portugese (Brazil)
    Copy coordinates  Koordinaten kopieren
        German, Portugese (Brazil)
    Maps  Karten
        German

Create beacons

    Invalid name  Ungültiger Name
        German, Portugese (Brazil)
    Invalid latitude  
        Portugese (Brazil)
    Invalid longitude 
        Portugese (Brazil)
    Invalid elevation
        Portugese (Brazil)
    Elevation feet (Optional)
        Portugese (Brazil)
    Elevation meters (Optional)
        Portugese (Brazil)
    Notes (Optional)
        Portugese (Brazil)

Weather

    %.0f °F
    %.0f °C
    %.0f%% humidity   Luftfeuchtigkeit
    No humidity data  Keine Luftfeuchigkeitswerte
    Dew point %s    Taupunkt
    Monitoring weather in the background  Wetterbeobachtung im Hintergrund
        German, Portugese (Brazil)
    Storm Alert  Sturmwarnung
        German, Portugese (Brazil)
    A storm might be approaching   Ein Sturm kann sich nähern
        German, Portugese (Brazil)
    Storm alerts  Sturmwarnungen
        German, Portugese (Brazil)
    Alerts  Warnungen
        German, Portugese (Brazil)
    %s %s - %s ago
        Portugese (Brazil)
    Set: %s - %s ago
    %s / 3h
        Portugese (Brazil)
    in
        As in the unit inches  In Incheinheiten
        German, Portugese (Brazil)
    Heat danger  Hitzegefahr
    Heat exhaustion/stroke risk. Be sure to stay hydrated, stay indoors or in shade, limit exercise, and watch for signs of heat exhaustion or stroke.
    Hypothermia/frostbite risk. Be sure to stay hydrated, dress warm and limit exposed skin, and watch for signs of hypothermia or frostbite.
    Hitze Erschöpfungs/Schlaganfallrisiko. Trinken Sie genügend, halten Sie sich drinnen oder im Schatten auf, limitieren Sie ihre Übungen und achten Sie
    auf Anzeichen von Erschöpfung oder Schlaganfall.
    Unterkühlung/Erfrierungsrisiko. Trinken Sie genügend, kleiden Sie sich warm und vermeiden Sie unbedeckte Haut. Achten Sie auf Anzeichen von
    Unterkühlung oder Erfrierung.
    
    
    Heat alert  Hitzealarm
    Heat warning  Hitzewarnung
    Heat caution  Hitzewarnung
    No temperature warning
    Hypothermia caution Unterkühlungswarnung
    Hypothermia warning  Unterkühlungswarnung
    Hypothermia danger Unterkühlungsgefahr
Temperature accuracy is affected by your phone's heat.
Die Genauigkeit der Temperatur wid von der Wärme ihres Telefons beeinflusst.

kylecorry31, 2020-09-24:
@raphaelventura thanks, I'll check it out!

@paulle thanks for the translations, I'll add them today

aw-bib, 2020-09-24:
Dialogs

  -  Copied to clipboard
     - German: `In Zwischenablage kopieren`

Disclaimer

   - Trail Sense is a tool that can be used in a survival situation, but you should not rely solely upon it. Please validate it's accuracy, avoid dangerous situations when possible, stay calm, and use common sense.

    - German:
      ```Trail Sense ist eine Anwendung die in
      Überlebenssituationen verwendet werden kann, aber Sie sollten
      sich nicht ausschließlich auf sie verlassen. Überprüfen Sie
      die ausgegebenen Werte, vermeiden Sie gefährliche Situationen
      wo immer möglich, bewahren Sie Ruhe und nutzen Sie den
      Gesunden Menschenverstand.```

Accuracy

    - Tips for improving accuracy
        - German: Tipps um die Genauigkeit zu verbessern

    GPS: Go outside with an unobstructed view of the sky.
        - German: `GPS: gehen Sie nach draußen mit einem freien Blick auf den Himmel`
    %s of actual location
        - German: `%s der aktuellen Position` (may depend on %s)
    %s of actual altitude
        - German: `%s der aktuellen Höhe`
    %s satellites
        - German: `%s Satelliten`
    Compass: Rotate device in a figure 8 pattern and move away from metal.
        Portugese (Brazil)
        - German: `Entfernen Sie sich von Metallgegenständen und rotieren Sie das Gerät in Form einer 8`
    GPS lock acquired
        - German: `GPS Fix vorhanden`

Flashlight

    - Flashlight
        - German: `Taschenlampe`
    - Tap to turn off
        - German: `Tippen zum Abschlaten`
    - SOS
        - German: `Notsignal / SOS`

Destination panel

    - ~ %s away
        - German: `~ %s entfernt`

Beacon list

    - Edit beacon
        -Long press a beacon to view options.
            - German: `Wegpunkt lang drücken um Optionen anzuzeigen`
Create beacons

    - Invalid latitude
        - German: `Ungültige geographische Breite`
    Invalid longitude
        - German: `Ungültige geographische Länge`
    - Invalid elevation
        - German: `Ungültige Höhenangabe`
    - Elevation feet (Optional)
        - German: `Höhe in Fuß (optional)`
    - Elevation meters (Optional)
        - German: `Höhe in Metern (optional)`
    - Notes (Optional)
        - German: `Anmerkungen (optional)`

Weather

    - %.0f °F
        - German: `%.0f °F`
    - %.0f °C
        - German: `%.0f °C`
    - %.0f%% humidity
        - German: `%0f%% rel. Feuchtigkeit`
    - No humidity data
        - German: `Keine Daten zur Luftfeuchte`
    - Dew point %s
        - German: `Taupunkt %s`
        (I guess the unit is attached based on the settings used)
    - %s %s - %s ago
        - German: `%s %s - %s früher`
        (this is hard to translate without context)
    - Set: %s - %s ago
        - German: `Setze %s %s - %s früher`
        (this is hard to translate without context)
    - %s / 3h
        - German: `%s / 3h`
    - Heat danger
        - German: `Hitzegefahr`
    - Heat exhaustion/stroke risk. Be sure to stay hydrated, stay indoors or in shade, limit exercise, and watch for signs of heat exhaustion or stroke.
        - German:
        ```Gefahr durch Hitzeerschöpfung/Hitzschlag. Trinken
          Sie genug, halten Sie sich nicht unnötig im Freien auf,
          bleiben Sie in Gebäuden oder im Schatten. Vermeiden Sie
          körperliche Anstrenungen und Sport. Achten Sie auf Zeichen
          von Erschöpfung oder Hitzschlag.```
    - Hypothermia/frostbite risk. Be sure to stay hydrated, dress warm and limit exposed skin, and watch for signs of hypothermia or frostbite.
        - German:
        ```Gefahr durch Unterkühlung/Erfrierungen. Trinken Sie genug,
        ziehen Sie sich warm an und bedecken Sie möglichst die Haut.
        Achten Sie auf Zeichen von Unterkühlung und Erfrierungen.```
    - Heat alert
        - German: `Alarm: Hitze`
    - Heat warning
        - German: `Warnung: Hitze`
    - Heat caution
        - German: `Vorsicht: Hitze` (? may depend on context. If it's
          some sort of level 1,2,3 with the above, translation is ok.)
    - No temperature warning
        - German: `Keine Temeraturwarnungen`
    - Hypothermia caution
        - German: `Vorsicht: Unterkühlung` (see heat)
    - Hypothermia warning
        - German: `Warnung: Unterkühlung`
    - Hypothermia danger
        - German: `Alarm: Unterkühlung`
    - Temperature accuracy is affected by your phone's heat.
        - German: `Die Genauigkeit der Temperaturangabe hängt von der Wärme Ihres Geräts ab.`

raphaelventura, 2020-09-24:
@kylecorry31, just tried transifex for openstreetmap translations. Very efficient and simple to use as well on the translator side, although not libre ;)

kylecorry31, 2020-09-24:
@aw-bib and @paulle does "Gefahr: Hitze" and "Gefahr: Unterkühlung" also work for heat danger and hypothermia danger?

@aw-bib yes, those are a scale, ex.
1. Heat caution
2. Heat warning
3. Heat alert
4. Heat danger

kylecorry31, 2020-09-24:
For the "Set: %s - %s ago" and "%s - %s ago" those refer to the pressure reading under the pressure chart when you tap on it (so it will say for example: 1000 hPa - 1h ago)

The one that starts with Set: indicates the set pressure (so it will say: Set: 1000 hPa - 1h ago).

Does that change the translation?

paulle, 2020-09-24:
Yes

-------- Original-Nachricht --------

kylecorry31, 2020-09-24:
@raphaelventura looks like Weblate does have free hosting for open source projects - I put in a request for the Libre plan. I got it setup (still in the trial) and I'm definitely still learning how to use it, but it seems to be working right now with GitHub: https://hosted.weblate.org/projects/trail-sense/trail-sense-android/

raphaelventura, 2020-09-24:
Nice ! I've spent some time translating and suggesting small changes in the source strings, will complete the work soon.

kylecorry31, 2020-09-24:
Translations have moved to Weblate (@aw-bib and @paulle I already moved your provided translations over to the source code, so no need to reenter them - thanks!). See the link in the issue description.

raphaelventura, 2020-09-25:
Hi @kylecorry31 ,

I have finished a first round of translation to french through weblate. Thanks for making the tool available so quickly !

The current debug version systematically crashes on my phone, so unfortunately I can't get context which makes translating harder and I had to guess some stuff using, e.g., the strings keys. Nevertheless, here are a couple of context-related questions that might help me improve my current translations:

- What is the legacy compass ? Are there multiple sensors that can be leveraged as compasses inside a phone ?
- I'm not sure how to handle the 'Degrees Decimal Minutes' geographical coordinates unit. Is it some kind of official unit ? This looks to me as two standard units mixed together (decimal and sexagesimal)
- Could you provide some context for the following 'override' values ? Specifically, are they overridden by other sensors values or entered manually ? Are they switches or editable fields ?
  - `compass_from_gps`
  - `compass_declination_override`
  - `declination_override_updated_toast`
  - `pref_latitude_override_title`
  - `pref_longitude_override_title`
  - `pref_gps_override_btn_title`
  - `altitude_from_gps`
  - `altitude_from_pressure`
  - `pref_altitude_override_title`
  - `altitude_override_updated_toast`
  - `pref_altitude_override_sea_level_dialog_msg`
  - `pref_altitude_override_sea_level_dialog_title`
  - `pref_altitude_override_sea_level_title`
- (Clarification + suggestion) For the `temperature_disclaimer` string, is it really true that the Temperature *accuracy* is affected by the phone's heat ? Or is it the *sensed value* that may be erroneous (positively biased) because of the phone's high temperature ? Accuracy is different from a systematic error !
- `pref_adjust_for_temperature_title` and `pref_sea_level_use_rapid_title` : what does 'Factor' mean here ? Is it a mathematical factor used for a proportional conversion from sea level pressure conversion ? Could you provide some info about these two conversions ?

kylecorry31, 2020-09-25:
@raphaelventura thanks! It's concerning that the debug build isn't working, would you mind trying this one out? https://www.dropbox.com/home/Trail%20Sense%20APKs I put some things in place which may fix the crash.

- The legacy compass is another sensor (software-based) available in some phones which Google says shouldn't be used anymore (but it may work better on some lower-end devices)
- Degrees Decimal Minute is an official format for coordinates, ex. X° X.XXX' N   - not sure how often it's used though (and yes, it's a combination of the two standard units)
- Most of those values are for "buttons" which will replace the override with the value from that sensor
    - compass_from_gps: will use your current location to set the declination override
    - compass_declination_override: The title of the declination override for the compass (edit text field)
    - declination_override_updated_toast: A message which appears when the declination is updated from the GPS on the calibration page
    - pref_latitude_override_title: The title for the edit text field the user can supply their current latitude (basically mock GPS)
    - pref_longitude_override_title: Same as latitude
    - pref_gps_override_btn_title: The button which can be used to update the lat/lng overrides from your current location
    - altitude_from_gps: The button which can be used to set the current altitude override from the GPS sensor
    - altitude_from_pressure: The button which can be used to set the current altitude override from the barometer
    - pref_altitude_override_title: The title for the edit text field where the user can enter their current altitude (a manual override)
    - altitude_override_updated_toast: The popup which appears when the altitude override is updated from the GPS or barometer
    - pref_altitude_override_sea_level_dialog_msg: The content of the edit text dialog which pops up when the user clicks the set override from barometer button (asks user to enter the sea level pressure)
    - pref_altitude_override_sea_level_dialog_title: The title of the edit text dialog above
    - pref_altitude_override_sea_level_title: The supporting text for the altitude_from_pressure button
- You are right, the temperature may be erroneous - poor wording on my part, I'll update the English string for that
- Factor implies they will be used to convert the value to sea level (mathematically) - the temperature is used in a mathematical formula while the rapid changes will be used to ignore noisy GPS readings

I'll add these to Weblate later.

raphaelventura, 2020-09-28:
Hi @kylecorry31,

The link you provided for the debug apk asks for a login, are you sure it's the right one ?

Thanks for all the details, I'll update my translations accordingly ASAP.

kylecorry31, 2020-09-28:
Thanks - yeah, I didn't realize that link would ask for a login - there's another sharable link which doesn't (on here https://github.com/kylecorry31/Trail-Sense/issues/74) 

But as you mentioned, there's a new release which is more up to date.

raphaelventura, 2020-09-28:
Hi again,

I've done a second round of translations and improvements based on your inputs. I can't see the weather-related stuff inside the app, but I guess this is the expected behavior when you don't have the mandatory sensors.

Here are two questions:
- I can't find 'Moon azimuth' and 'Sun Azimuth' in the strings to be translated but maybe I didn't search well enough.
- Some of my translated option titles are too long to be printed on my screen and therefore end with an ellipsis sign. Is there any way to force a line break or should I try to shorten the titles ?

And some suggestions:
- There should be some way to switch between upper and lower case after the "until" string in the "Astrology" tab in order to print something like "until sunset" and not "until Sunset".
- I'd like to suggest a little refactoring for the location permission 'onboarding' screen : you could merge all the strings detailing why this permission is needed into one e.g. changing
```
 Allows for more accurate weather forecasts (adjusts for altitude)
 Allows you to navigate to beacons
 Allows you to see sun/moon rise and set times
```
to
```
Giving the app permission to access location will  :
- Increase weather forecasts accuracy (by leveraging altitude info)
- Allow you to navigate to beacons
- Inform you of sun/moon rise and set times
```
and keep the following string regarding privacy and 'full permission' separated.

kylecorry31, 2020-09-28:
@raphaelventura Thanks! 

- I must have missed those (sun and moon azimuth), just pushed them up
- Can you send a screenshot of when the titles overflow and I'll investigate some options?
- As for the capitalization of the sunset/sunrise there, yeah I can look into a way to allow both
- That refactor sounds good! I can update it later today

raphaelventura, 2020-09-28:
Great ! I've just completed the translations.

Here's a [screenshot](https://upload.disroot.org/r/Q2NZ_Gux#BBRTAZe7iYNozzkrGupu9KlYP7obpV24W66tY1DSsII=) where you can see an example of text overflowing.

paulle, 2020-09-28:
I believe a lot of translations from English to another language will have more letters. At least I know it for sure when it comes to English German translation. So the the English version should have an option for longer text, line break for example.

-------- Original-Nachricht --------

kylecorry31, 2020-09-28:
Thanks, yeah I can look at some overflow options there. I would think that a line break would work (maybe \n in webflow, not completely sure). I'll see if there any settings I need to enable on those preferences to support that.

kylecorry31, 2020-09-28:
Okay, I updated the preferences to wrap instead of cutting off long text - no changes in Weblate needed. Thanks for all those new translations!

Leopardus4, 2021-01-05:
Hi! I have few thoughts about translations. Trail Sense uses terminology from astronomy/geography and has few settings which should be simple but understandable. But it's sometimes tricky to find good translation and/or understand impact of setting. There were questions in this thread and in #164 - and I guess there will be few in future. To simplify further translations (there are 25 languages at weblate!) I suggest considering and discussing few things:

- Maybe creating dedicated threads (issues) for specific parts of app. I. e. one general issue (this?) with links to other. Other: one issue for navigation, one for astronomy; settings; tools; guides; etc.. Each issue would contain description about given part (first post, updated from time to time?) and questions dedicated to given part. Few additional ideas below. It may be more clear for another translators to find proper informations this way.

- Few screenshots. It would help to get context of a phrase - especially for new features. Weblate allows to add them ([info](https://docs.weblate.org/en/latest/admin/translating.html#visual-context-for-strings)).

- Wikipedia links. Technical terms are sometimes hard to explain and hard to translate. Wiki provides good explanation and 'other languages' section - it would be helpful. Example: _This setting will allow to choose [type](https://en.wikipedia.org/wiki/Twilight) of dusk/twiling_

- I think it's worth to consider where to put further explanations & questions - github or weblate - and choose one place. Both support screenshots, additional explaination ([link](https://docs.weblate.org/en/latest/admin/translating.html#additional-info-on-source-strings)), but i. e. at weblate discussion is probably rather centred on single phrases :unamused:.

And one question: why is [this](https://hosted.weblate.org/translate/trail-sense/trail-sense-android/en/?checksum=253f0e3ce8988080#translations) phrase locked as read-only in non-english languages? I guess it's a text in settings > compass, below _true north_ option.

kylecorry31, 2021-01-05:
@Leopardus4 Thanks for the suggestions, I will look into some options! GitHub recently released a new discussions feature as well, which might help out. 

As for that Declination phrase, when I look at it in weblate, it doesn't appear to be locked, can you try it again? (and yes, that is the correct text)

Leopardus4, 2021-01-05:
Thank you for reply.

Yes, this phrase is locked for me. _English_ page - not locked, but any other - marked as _read only_, I can't enter any text in _translation_ field nor save/suggest it.

kylecorry31, 2021-01-06:
@Leopardus4 oops, I saw the Edit button and assumed that I could edit for other languages - didn't realize there was a read-only icon on that. Not sure why that happened, but I was able to resolve it (I think) - can you check again?

Leopardus4, 2021-01-06:
Yup, it works now. Thanks :)

hphan9, 2021-10-17:
Hi, Can I help translate it to Vietnamese?

hphan9, 2021-10-31:
Hi @kylecorry31, I have translated around 50 words in Vietnamese. However, I did not see it updated on my Hactoberfest profile, (https://hacktoberfest.digitalocean.com/) . Could you please check ? Thank you

kylecorry31, 2021-10-31:
@hphan9 It appears that everything is set up correctly in this repo, and I also added the label to the PRs that I merged in (which showed you as a contributor). I'm thinking the hacktoberfest side of things isn't detecting it because it was done through Weblate. Can you create a PR on this repo with a trivial change (like adding a new line to the readme or something)? 

That might force it to identify you as a contributor

hphan9, 2021-11-01:
@kylecorry31  Thank you for your response. It is ok now

kylecorry31, 2025-12-01:
> [@kylecorry31](https://github.com/kylecorry31) shouldn't "_white_" be "_with_" in these entries?
> 
> [Trail-Sense/guides/en-US/guide_survival_chapter_medical.txt](https://github.com/kylecorry31/Trail-Sense/blob/ec203c86a6034952bba25a7349a168f74862afcb/guides/en-US/guide_survival_chapter_medical.txt#L800)
> 
> Line 800 in [ec203c8](/kylecorry31/Trail-Sense/commit/ec203c86a6034952bba25a7349a168f74862afcb)
> 
>  If you do not have soap, you can substitute white ashes, sand, or loamy soil. 
> [Trail-Sense/app/src/main/res/raw/guide_survival_chapter_medical.md](https://github.com/kylecorry31/Trail-Sense/blob/ec203c86a6034952bba25a7349a168f74862afcb/app/src/main/res/raw/guide_survival_chapter_medical.md#L800)
> 
> Line 800 in [ec203c8](/kylecorry31/Trail-Sense/commit/ec203c86a6034952bba25a7349a168f74862afcb)
> 
>  If you do not have soap, you can substitute white ashes, sand, or loamy soil. 
> and
> [Trail-Sense/site/src/survival-guide/guide_survival_chapter_medical.md](https://github.com/kylecorry31/Trail-Sense/blob/ec203c86a6034952bba25a7349a168f74862afcb/site/src/survival-guide/guide_survival_chapter_medical.md?plain=1#L844)
> 
> Line 844 in [ec203c8](/kylecorry31/Trail-Sense/commit/ec203c86a6034952bba25a7349a168f74862afcb)
> 
>  If you do not have soap, you can substitute white ashes, sand, or loamy soil.

Hi, no it should be "white ashes" - though if needed in a translation the word "with" can be added after "substitute" (I believe the grammar is correct with or without "with" there).