kylecorry31/Trail-Sense #214: Pressure tendency evaluation

aw-bib, 2020-09-24:
Just noticed this today as the cold front of some storm depression on the North sea was passing during the night, as one can also see nicely in the barograph.

As you can see in the screenshot Trail Sense gives an `improving soon` based on the `1.9hPa/3h` increase. However, the increase  over the last 3h was more like `1002-997 = 5 hPa` which conforms a bit better with the current wind situation and current coastal _strong wind warning_ for the Elbe (Hamburg-Cuxhaven 5-6 [Bft](https://en.wikipedia.org/wiki/Beaufort_scale), squalls up to 7 Bft) and measurements at a nearby meteorological station of 5 Bft (~16 kts).

So, I wonder if the tendency is calculated not for 3 real hours (if available) thus averaging, but just as the gradient between the last two points measured. Then it may fall in this situation for the slightly flatter increase in the last measurements and _under-estimate_ the situation. (This would be erring on the dangerous side.)

Here's the screenshot from Trail Sense from 11:14 CEST (UTC+2)

[image]

And just for comparison the relevant part of the analysis chart from Deutscher Wetterdienst, issued Thu, 2020-09-24, 06:00 UTC (isobars are 5hPa apart, the low (marked `T`) is moving in NE direction):

[image]

isRelatedTo: #179

kylecorry31, 2020-09-25:
Thanks for the details - I'll look into why the tendency differed. 

The current formula is:
- Find the pressure reading which is closest to the 3h mark
- Take the last recorded pressure (from the history graph - not the one displayed at the top of the screen)
- Calculate the tendency from those two points

There may be several causes for why it's displaying 1.9 instead of 5 there:
- The pressure reading which is closest to the 3h mark could be displaying as 2h on the chart above depending on what your update interval is set as (I'll have to check again, but I think if it is 2h 35m it will display as 2h and ex. 3h 35m will display has 3h, so it will use the 2h reading in that case since it is closer to the 3h mark) - might be wrong there, I'll be sure to check that later today
- There could be some rounding issues in play - namely that the UI rounds to the nearest whole number (changing in 0.9.0 to 1 decimal place), though that can't account for a 3 hPa difference
- The readings being used are from the pressure history chart, not the one displayed on the top of the screen, so it may differ. This was done a while back because when the current reading was used the forecast could change rapidly when the tendency was close to the improving threshold (for example) - maybe it should be using the current pressure - thoughts?

**For what it's worth, I just tested on 0.9.0 and the calculation looks fine when using the pressure graph's 3h and last reading (comes out to 1.3 hPa, which is what I see in the UI). When using the current pressure, I calculate a -2.1 reading... so I think this might be related to the sea level pressure conversion algorithm (issue #194) - I don't see a discrepancy when turning off sea level pressure.**

kylecorry31, 2020-09-25:
Also, the focus of my next development cycle will be stability and accuracy (for the 1.0.0 release) - I'll include this case and will be trying to improve the sea level pressure conversions.

aw-bib, 2020-09-26:
> pressure reading which is closest to the 3h mark could be displaying as 2h

There might be some issue indeed as I only get full hours form a certain point onward. I tried to be careful to hit the first "3h reading", but my fingers might be a limiting factor. Some stepping buttons (forward, backward) come to mind to "play" trough the graph, but except for debugging they might be of limited use.

> so I think this might be related to the sea level pressure conversion algorithm

I have a gut feeling as well that there might be something buried there, indeed. Today, we have a completely different picture. The barograph shows an almost horizontal line, only minor changes. If I correct for sea level pressure however, it actually _is_ a straight line, no bumps what so ever, they all get smoothed out.

> I don't see a discrepancy when turning off sea level pressure.

I'll give that a try for the time being. It should give the proper tendencies and differences if I stay on the same height, which is quite easy here in northern Germany. (More of a challenge would be to find a moumtain to drastically change height ;)

> I should also point out that unless you have drastically changed altitude, the tendency displayed in the UI when sea level pressure is enabled should be very close to the one when sea level isn't enabled

Of course it should, but I fear this is not the case. Even yesterday, we had another front passing slightly after the one in the screen shot and if I corrected for sea level pressure I got a _straight, horizontal_ line. Looks like something is fishy there. I'll observe this a bit more.

aw-bib, 2020-09-27:
Again a passing front during the night.

Without correction for sea level:

[image]

Switch on the correction for sea level by given altitude:

[image]

kylecorry31, 2020-09-27:
Thanks for the information, I hope to spend a decent amount of time reviewing the formula this week

aw-bib, 2020-10-18:
Reporting only quite late as 0.9.2 just hit F-Droid after two weeks. AFAIS the algorithm done in #194 really improves sea level correction :) Right now it is almost in perfect agreement with the weather station.

I just noticed this yesterday, that I don't get a graph any more, it display pressure, icon, textual evaluation ("improving soon") and the graphs headline. Wondering what's going on I _disabled_ sea level correction and got this:

[image]

The marked spike, basically to the vacuum of space, stems from a reboot. Maybe the graph gets lost due to some "division by zero"-except(?)

kylecorry31, 2020-10-18:
Thanks! I believe I have that fixed in the latest version, I don't think it's the one posted on Dropbox though.

aw-bib, 2020-10-19:
> fixed in the latest version

Ok, I'll watch out once F-Droid is updating again :)

closed 2020-10-23 by kylecorry31

aw-bib, 2020-10-27:
Seems to be fixed indeed. Thank you :)