kylecorry31/Trail-Sense #221: Companion: TS Augmented Reality

kylecorry31, 2020-09-27:
A companion app idea that will display beacons + sun and moon paths as an overlay of the camera. There could be a button in trail sense to open a beacon in AR navigation / or a tool item to launch it (just like there will be for TS Maps).

This would be a separate app because I don't really want TS to require camera permission + this feature may not be really useful and just add extra bloat to TS (but it would be cool, and something I want to explore).

kylecorry31, 2020-11-20:
- Viewing nearby beacons (with altitude)
- View the path of the sun and moon throughout the day
- Cloud identification (can help determine weather patterns)
- Inclinometer (object height measurement)

kylecorry31, 2020-11-21:
[image]

Android actually makes this pretty easy to implement - needs some UX work, but hopefully I'll get it in place by the end of the year.

qwerty287, 2021-04-13:
Hello,
I just reimplemented this feature into the latest version/commit of TS and added an option in the settings to disable/enable AR background. Are you interested in adding my code to the AR Branch?

kylecorry31, 2021-04-13:
Awesome, I'll take a look! Can you send the link? I still haven't entirety figured out how I want to implement AR or if I do, but would definitely be interested

kylecorry31, 2021-04-13:
As you can tell from the demo I showed above, it didn't really mesh with the UI well, and I was having trouble getting the azimuth to line up with real world angles (i.e. making the compass drawn on the inside of a sphere)

qwerty287, 2021-04-14:
I created a fork, but for now, I only implemented the setting and the camera background and there are some problems (with the UI, I didn't fix them). https://github.com/qwerty287/Trail-Sense

qwerty287, 2021-04-14:
I created a fork, but for now, I only implemented the setting and the camera background and there are some problems (with the UI, I didn't fix them). Unfortunately, I couldn't test it with my real smartphone (#551), but in an AVD, it worked.
 https://github.com/qwerty287/Trail-Sense

kylecorry31, 2021-04-14:
Thanks! I think there is still plenty of work before this could be released. I was more thinking of having a viewfinder button on the linear compass, as the use case would be marking a destination bearing with better accuracy - and not having the camera always on/full screen - that might solve the sphere projection issue as well (or avoid it).

qwerty287, 2021-04-14:
Ok, I'll see how I implement this.

kylecorry31, 2021-04-14:
Awesome thanks, if you want I can upload a hand drawn mockup

qwerty287, 2021-04-14:
Yes, would be great if you upload your mockup.

kylecorry31, 2021-04-14:
[image]

The text says:
Viewfinder height = height of round compass

Click to toggle camera, default off
(first click requests permission - you can ignore this part, I have some other ideas on camera permissions since that is a sensitive permission)

Essentially the viewfinder will have a thin line (1 - 2 dp) down the center which is behind the linear compass. You can tap it to set a destination bearing. The camera image shown should be cropped to fit, not scaled, and I don't have a strong preference for the width as long as it isn't full screen (maybe 100 - 200 dp would work)

The rest of the UI can stay the same, as this would avoid having text overlays.

kylecorry31, 2021-04-14:
My only outstanding concern with the augmented reality is that the CameraX library does not have a stable release yet, and the camera view is in alpha - though that most likely just means that this feature will be released behind the experimental setting at first.

qwerty287, 2021-04-14:
I implemented it like the mockup (the thin line is missing and the image is scaled, I have to fix this)
https://github.com/qwerty287/Trail-Sense/commit/9d5e33b07787a3e55675f7b66b046a6e3513a70a

kylecorry31, 2021-04-14:
Awesome! I'll take a look later today, and create a branch that you can do a pull request to if you want

kylecorry31, 2021-04-14:
I created a new branch called Sighting-Compass - feel free to create a pull request there

kylecorry31, 2021-04-15:
I just ran the branch and everything looks good, great work!

A few recommendations:
- Instead of clicking the line to place a destination bearing, allow clicking the camera view itself (I found it hard to actually get the click event to trigger on the sighting line)
- Remember if the user clicked the camera button and turn it on automatically if they re-enter the linear compass mode
- If cropping is difficult, you could just show the full image as you have now (possibly sizing the box to the aspect ratio of the image to avoid the black bars on the side)   -  I wouldn't spend too much time here, because I eventually plan on making a custom cameraview using my sensor framework with the camera, that should allow easier cropping

kylecorry31, 2021-04-15:
I've also updated the description of this issue to talk about how I plan on handling the permissions with this if you are interested. 

Because of the work you did to get this running, I think this will be good to include in the 2.2.0 release (probably not for another month - I'm still working on the 2.1.0 features)

qwerty287, 2021-04-15:
> I just ran the branch and everything looks good, great work!
>
> A few recommendations:
>
>     * Instead of clicking the line to place a destination bearing, allow clicking the camera view itself (I [...]

Thanks! I'll implement this.

>     * Viewing nearby beacons (with altitude)
> 
>     * View the path of the sun and moon throughout the day
> 
>     * Cloud identification (can help determine weather patterns) - can be done with a user uploaded image
> 
>     * Inclinometer (object height measurement)

These features shouldn't be implemented, right?

kylecorry31, 2021-04-15:
Thank you!

And no, those features should not be implemented just yet - several of them require drawing on the inside of a sphere with the camera at the center, from my experimentation on it it was difficult to do

qwerty287, 2021-04-15:
I implemented the recommendations (better size of camera view, toggle destination bearing with the camera view and don't close camera view if linear compass mode is left and entered again).
#560 

But I found some problems:
- If you opened the camera once, it runs always in the background, even if the camera view is hidden.
- As a result of this, you can't use the flashlight (via a quick action) if you opened it once
- If you turn on the flashlight and then open the camera view, the flashlight is turned off, but it looks like it's on. 
- image is scaled, not cropped
- permission request appears always if you open the app, even if experimental features are disabled
The two problems at the beginning are a result of the "bindToLifecycle" function, but I don't know how to start the camera, but end it if you leave camera view.
By the way, Google says this about the camera view:
"CameraView is being deprecated" (https://developer.android.com/jetpack/androidx/releases/camera#extensions-view-1.0.0-alpha23)
Maybe I see if I can migrate this Camera View to the newer view.

kylecorry31, 2021-04-15:
Thanks! I would say yes to switching to that new camera view, https://developer.android.com/reference/androidx/camera/view/LifecycleCameraController (I saw the message in my email but not on GitHub)

It looks like the unbind() method would resolve the staying open in the background issue. A temporary solution with the current CameraView would be to pass null to bindToLifecycle, that unbinds it and stops the camera

And as for the flashlight, you might be able to use this to resolve that (it will also cancel the notification):
val handler = FlashlightHandler.getInstance(context)
handler.off()

qwerty287, 2021-04-16:
> Thanks! I would say yes to switching to that new camera view, https://developer.android.com/reference/androidx/camera/view/LifecycleCameraController (I saw the message in my email but not on [...]

Thanks for the solutions! I implemented those from the pull request already, and I'll add these too.

kylecorry31, 2021-04-17:
FYI, I merged main into the sighting-compass branch and updated the camera library version to use the latest available.

qwerty287, 2021-04-17:
Ok, thanks!

qwerty287, 2021-04-20:
The problem with the cropped image could be fixed by set the width to (almost) fullscreen. For me, this wouldn't be a problem. (I found another way to crop the image, but this didn't work well. The cropped image was from the top left corner of the image, and not from the center. And the image quality was very bad.)
In an AVD, this is the result:
[image]

kylecorry31, 2021-04-20:
I'm not sure I would want it full screen, because that would tend to imply that the tick lines match up to the camera image (so there would be a weird parallax effect when moving)

I am actually fine with the way it is currently (the small image, not cropped) - maybe what could work is specifying a zoom level on the camera (like 1.5x or 2x zoom). That would allow a more zoomed-in image ("center cropped") while retaining the aspect ratio.

qwerty287, 2021-04-21:
I wouldn't move the SeekBar to the settings, it's something very useful, I think. Maybe, you could hide it and view it with another button or another action by the user.

kylecorry31, 2021-04-24:
Thanks for your work on this! I changed the permission workflow to only request the camera permission when you activate it in settings to better request user privacy around using the camera.

Everything seemed to be working fine, I may make the seekbar have more tick values on it to allow for smoother zooming.

closed 2021-04-24 by kylecorry31