kylecorry31/Trail-Sense #224: Clicking on settings tab crashes the app

swedneck, 2020-09-27:
using release 0.9.0
on AOSP Extended, android 8.1.0

kylecorry31, 2020-09-27:
@swedneck did you download the APK from the releases tab on GitHub?

swedneck, 2020-09-27:
Yes

kylecorry31, 2020-09-27:
Okay, I'm going to see if I can replicate it on an emulator - I'm not having any luck replicating on my phone

kylecorry31, 2020-09-27:
@swedneck no luck replicating with a fresh install of that APK.

What language is your phone set to? 

I don't get any logs from crashes on non-Google Play Store installs, so it's a bit hard to track it down.

swedneck, 2020-09-27:
English and swedish, i think?
Don't remember which version i upgraded from, nor if i got it from github or fdroid

kylecorry31, 2020-09-27:
Okay, thanks. 

Can you install this APK? https://www.dropbox.com/home/Trail%20Sense%20APKs

I added some error catching logic, which will display a popup with an "Email developer" button. Can you use that to send me an email of the error when it happens?

swedneck, 2020-09-27:
I need a Dropbox account to download that, apparently

kylecorry31, 2020-09-27:
Hmm, I wasn't aware of that. Here's the APK direct link: https://kylecorry.com/assets/Trail-Sense.apk

swedneck, 2020-09-27:
Email sent

kylecorry31, 2020-09-27:
Thanks, I know what the issue is now. I'll be rolling out a hotfix soon - thanks for running that debug build for me!

kylecorry31, 2020-09-27:
New release: https://github.com/kylecorry31/Trail-Sense/releases/tag/0.9.1

closed 2020-09-27 by kylecorry31