kylecorry31/Trail-Sense #225: App crasher at first launch when location is disabled

raphaelventura, 2020-09-28:
Backtrace follows :

```
Version: 0.9.1
Device: OnePlus bacon (A0001)
Android SDK: 29
Message: abstract method "void android.location.LocationListener.onProviderDisabled(java.lang.String)"

java.lang.AbstractMethodError: abstract method "void android.location.LocationListener.onProviderDisabled(java.lang.String)"
at android.location.LocationManager$ListenerTransport._handleMessage(LocationManager.java:365)
at android.location.LocationManager$ListenerTransport.access$000(LocationManager.java:275)
at android.location.LocationManager$ListenerTransport$1.handleMessage(LocationManager.java:291)
at android.os.Handler.dispatchMessage(Handler.java:107)
at android.os.Looper.loop(Looper.java:214)
at android.app.ActivityThread.main(ActivityThread.java:7356)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:940)

```

kylecorry31, 2020-09-28:
Thanks. This is odd, I wouldn't have expected that to happen, nor can I reproduce the issue. I do have an idea of how to fix that and created a new debug build, could you test that out for me? https://www.dropbox.com/sh/gplqslml3iyq2ec/AACn0Arv3KRHAUF1wCusmoZ2a?dl=0

raphaelventura, 2020-09-28:
No crash with this build !

kylecorry31, 2020-09-28:
Yay, I'll do a hotfix in a few hours. Thanks!

closed 2020-09-28 by kylecorry31