kylecorry31/Trail-Sense #241: Show pressure graph on barometer calibration

kylecorry31, 2020-10-10:
This will help with sea level pressure calibration parameters, as it can update as the settings are changed

aw-bib, 2021-01-02:
I still don't get how to set the parameters used for sea level pressure calibration properly. Maybe this could come to rescue here?

E.g. for one of my usual walks I can predict that _a storm will come in_ as soon as I am back home, simply as I get a spike once I walk down to the river and back up again. TS takes a measurement every 15min or so, so it "sees" these 2h walks. I don't mind this warning per see as a quick glance at the curve gives a pretty simple explanation. I just wondered if the new algorithm shouldn't "see" this as an actual height change. So, I think I will have to change the correction parameters, but  how do I get an idea which one and how much? Is there some documentation already that I missed?

Here is an example reading:

[image]

The spike at the beginning of the curve was my walk. This one is steep enough to trigger the storm warning (> 3 hPa/3h). The fall that happened after the small spike then was a real storm coming in that triggered a (correct!) warning as well.

There seems also to be some influence from the Altimeter, I guess and the (auto) calibration done there. But this is also somewhat mysterious to me. Say, I switch of the automatism and use a known height or a known sea level pressure I would _expect_ that in both cases I get a sea level reading in the barometer that gives just the sea level pressure I set at this point (explicitly or implicitly). However, here it is always (quite) a bit off. E.g. right now I set 1012 -> 1018.5hPa, I set 23.5m -> 1018.5hPa. I guess this results from some history to be taken into account(?)

This would confirm my impression that the sea level reading drifts with time if there is no other input than the barometer. Could this be in principle?

IOW once #194 got implemented I took some measurements. I set the phone to the balcony and left it there for some hours. It had a clear sky, a GPS signal and in the end it was actually spot on. Now, I left the phone at home, same place, same height, though no GPS in my flat of course ;) I would expect it to track the weather station, which it does perfectly in tendencies (I do not get historic values from the station, unfortunately, so I can't compare actual readings). Looks all safe and sound. But if I compare the absolute values reported after some time they seem to drift apart with time. IOW I had some sky high readings on TS e.g. while we did have a H here but only a moderate one. (I don't see this in the `n hPa/3h`, but it could just be to slow for this.)

kylecorry31, 2021-01-02:
Yeah, I need to add some documentation on how to change those parameters - but I think this issue (the graph) will help out a lot. What I would recommend is to adjust the travelling altitude change threshold first until the spike is almost gone (which would mean you need to swap between settings and weather for now)  - for reference, I have mine set to 23 meters. You may also want to have the "ignore rapid altitude changes" enabled which will essentially stop readings while travelling to ignore spikes. The travelling pressure change threshold can be set to the maximum pressure change you see with storms in your area (which requires the rapid pressure change threshold setting to be enabled). I have mine set to 6 hPa / 3 hr. 

Yes, the value displayed there does take some history into account which factors in pressure changes and altitude changes. Something to note as well is that if you adjust the altitude (override) that will only affect readings from that point forward, and existing readings may still influence the currently displayed pressure (it will take 48 hours for the old readings to clear out). It essentially calculates the current altitude based on the average altitude of previous altitudes in which you haven't travelled (according to the values in settings), which allows it to ignore small spikes in GPS readings.

closed 2021-02-28 by kylecorry31

aw-bib, 2021-04-20:
> What I would recommend is to adjust the travelling altitude change threshold first until the spike is almost gone 

I fear I don't yet get _the right way to do it for the simple minded_. I think my main issue here is the "how". It seems that if I change the parameters, the _calculated_ display is not updated to reflect the results that these new parameters would have had, but it stays as done from the old set. Only measurements in the future are done by the new parameters set. This seems also to be true for the graph on the settings page. So it's not like I see the graph, change the slider(s) till the spike vanishes. I can adjust a bit, and next time I'll see how this affected the graph.

kylecorry31, 2021-04-20:
Hmm, I'll look into this more - when I adjust the parameters (travelling altitude/pressure change threshold), the graph updates immediately. Would you be able to post a screenshot of the barometer calibration page, and let me know what you have your traveling threshold parameters set to?

aw-bib, 2021-04-21:
> Hmm, I'll look into this more - when I adjust the parameters (travelling altitude/pressure change threshold), the graph updates immediately.

So I should be able to just adjust the sliders till the spike vanishes and I get a smooth curve, right?

> Would you be able to post a screenshot of the barometer calibration page

Sure enough, just took a dedicated stroll yestereve down to the river. According to the topographical map at hand it should be 40m -> 10m and back up again. This usually results in a storm warning as soon as I'm back home ;)

[image]

> what you have your traveling threshold parameters set to

Oriented it at the guidelines you [gave above](https://github.com/kylecorry31/Trail-Sense/issues/241#issuecomment-753545701) and then played around a bit over the last months. For the altitude change I could nail sort of a sweet spot around 25m (not perfectly smooth curve, but I did not manage a better one). I did not manage come to grips with the pressure change as I didn't see it's influence on the graph so it is sort of random. Currently, I came up with:

- Travelling altitude change threshold: 25m
- Ignore rapid altitude chagnes: ON
- Factor in raptid pressure changes: ON
- Travelling pressure changes: 4hPa/3h

Probably, I am already at the best spot.

What I can confirm however is, that I could move around both sliders as much as I want, the pressure curve on the settings screen stays the same, so the spike is always there and stays.

kylecorry31, 2021-04-21:
Yeah, it appears that is the best you'll get there. As you can tell, sea level calibration is hard to do with altitude from the GPS (as that is inherently inaccurate). The intention of the travelling thresholds were to only allow weather predictions while in the same location.

I do plan on revisiting this with the following open issue: #402

I'm planning on adding a location (rather than altitude) based traveling threshold. And tweaking some of the settings - unfortunately I can't find any research papers involving this issue, as weather barometers are really designed to be installed in one location.

aw-bib, 2021-04-21:
> Yeah, it appears that is the best you'll get there.

Ok. Thanks :)

> As you can tell, sea level calibration is hard to do with altitude from the GPS (as that is inherently inaccurate).

As you mention this: I did calibrate the barometer using a known pressure from an official station in the vicinity some time ago. Just as I do it with my weather station once I have to change the batteries. Since I did this, the barograph is _spot on_, almost to the decimals.

TS currently does _not_ auto calibrate for height changes. Maybe I'm wrong, but I understood the thresholds like the device actually sees the peak, but would level it out like "excluded as a measurement error" if the threshold is exceeded. But maybe I set it unintentionally to the "I nailed the device to the wall"-mode. IOW by using a barometric calibration, for height, did I actually _disable_ the thresholds through the backdoor? Probably my issue is just an issue on "Layer 8"?

> I'm planning on adding a location (rather than altitude) based traveling threshold.

Hm. Would be evaluating own movement relative to the "pressure cell". Say, if I move with the speed of a low in the direction of said low I would stay on the same pressure all the time and all barometric changes have to be height changes. Similarly, if I move upwards but am much slower than the depression all the effect goes to the latter. Similar to the evaluation of movement wrt movement of the depression as discussed in #71/ #101 to get an idea of the (gradient) wind speed. And it sounds at first glance a bit like "is my speed negligible to the speed of the weather system in place" where one could probably derive the latter from changes observed while not moving aka before the walk.

kylecorry31, 2021-04-21:
TS does auto calculate for height changes - if you have altimeter auto-calibration set to either GPS or GPS+Barometer. Give you said you calibrated to a known pressure, do you have altimeter auto-calibration set to never? If so, this is why you can't get the spike to go away as the altitude is manually overridden, therefore TS can't tell your altitude is changing (it is in "nailed on the wall mode").

Also, for reference, here's the raw altitude adjusted reading Trail Sense has to "smooth" out:
[image]

The noise in the graph all comes from the GPS altitude (barometer noise is minimal) - and TS only uses the GPS (technically the altimeter, so that includes overrides) to calibrate. The previous version of this algorithm used rapid pressure changes to calibrate but that led to ignored or inaccurate tendencies.

aw-bib, 2021-04-21:
> do you have altimeter auto-calibration set to never

Yes. You'd have spotted this immediately if I had posted the screeshot from the barometer page: `Set 990.1 hPa 940h ago`.

>  (it is in "nailed on the wall mode")

Thanks for this clarification!

I was imagining to adjust smoothing/fitting parametesrs for the barograph curve, instead I was adjusting calibration parameters for the disabled auto calibration.

May I suggest to disable these options in this case and give a hint for the "nailed to the wall" mode?

> So yeah, it's a bit tricky to get this right :)

Always was sure about this, and it does an amazing job.

kylecorry31, 2021-04-21:
Good idea, I should have disabled those, thanks!

commit 88e9c986 (2021-02-28): Add pressure chart to barometer settings

Lots of duplicated code here, clean that up later.

Closes #241