kylecorry31/Trail-Sense #244: Identify how to translate user guide

kylecorry31, 2020-10-14:
The .md files in the raw directory: https://github.com/kylecorry31/Trail-Sense/tree/master/app/src/main/res/raw

I believe you can create new directories for raw-lang with the translated files inside. I'm not sure at this moment how to get them into Weblate. Once I figure that out I can get them up there, in the meantime if anyone wants to provide translations directly, you can submit a pull request or just send me the translated .md file(s) and I'll add them.

qwerty287, 2021-07-07:
@kylecorry31 
Yes, you can create directories raw- + language code to translate them. 
Getting this on Weblate is not as easy as the app store listing (fastlane metadata, https://github.com/kylecorry31/Trail-Sense/issues/538#issuecomment-836659684), but you could take a look at (I don't know much about Weblate) https://docs.weblate.org/en/latest/formats.html#text-files. The site linked there says:
> The Translate Toolkit can process simple Text files. This is very useful for translating installation files and READMEs. The processing of these files is performed by the txt2po converter.

If it supports putting the translated files into the directories, this could work.

kylecorry31, 2023-02-02:
A Chinese translated user guide by Mellvie via email (this guide is likely to change before this issue is resolved, but it may be a good starting point). 
[photo maps.txt](https://github.com/kylecorry31/Trail-Sense/files/10568953/trans.txt)

kylecorry31, 2023-12-23:
The guides are a lot longer than regular text, what happens if it is only partially translated?

Maybe there should be a way to indicate that it is fully translated, or the guides can be automatically translated for the initial pass so that way they are complete.

Maybe add a button to switch to the English version of the guide in app?

kylecorry31, 2023-12-27:
It does not appear weblate supports translating markdown files in an android resource directory.

Options are:
1. Only allow translations via GitHub
2. Move all guides into a separate folder. That will contain subfolders for each guide which contain markdown files with the language (ex. guides/guide-tool-navigation/en.md). Then create a python script to convert these to raw resources. Downside is I'll need a component for each guide which doesn't sound sustainable.
3. Get weblate to recognize these files as app store metadata (it might require a specific format for that though) - maybe combine this with the separate guides folder above.

I will experiment with 3, and if that doesn't work I'll have to go with 1 until weblate has a better solution.

kylecorry31, 2023-12-30:
User guides can be translated here: https://hosted.weblate.org/projects/trail-sense/trail-sense-user-guide/

I need to add a build step to get them included in the actual APK, and I can add a check for completion there as well (ex. number of new lines is roughly equal).

closed 2024-01-01 by kylecorry31