kylecorry31/Trail-Sense #245: Add more user guides

kylecorry31, 2020-10-14:
Ideas:

Read the navigation tab UI
Create a beacon
Navigate to a beacon
Delete a beacon
Edit a beacon
Share a beacon
Create a beacon from a maps app
Display the ruler and adjust its scale
Turn on the flashlight / SOS
Check GPS accuracy
Calibrate compass
Tune compass settings
Tune GPS settings
Tune altitude settings

Monitor weather in the background
Convert readings to sea level
Change sensitivity
Receive storm alerts

Read the astronomy tab UI
Receive sunset alerts
Show the sun and moon on the compass

How to use the inclinometer

Navigating with a map and compass
Using the SOS flashlight mode to get help
Using the astronomy tab to plan
Using the inclinometer to determine avalanche risk

qwerty287, 2021-04-12:
I've written a simple guide for the water boil timer (at the first step you could write a little more, but I have no better idea):
```
Boil and purify water

## Boil and purify water so that it is safe to drink it
1. Open the Water Boil Timer
2. Filter the water
3. Bring water to a rolling boil
4. Start the Timer
5. When the timer has expired, enjoy your purified water!
```
(If you know better formulations, feel free to use them and feel free to correct my mistakes.)

And I have a question to the water boil tool, but this isn't the correct issue for this (maybe create another one): Why there's every time this rotating circle when opening the water boil tool? I don't think this tool needs to load much.
[image]

kylecorry31, 2021-04-12:
@qwerty287 Thanks! As for the loading indicator (I'll include that in the user guide), water boiling times depend on your altitude, so the indicator appears while Trail Sense obtains an altitude reading - though it probably can default to the last known altitude for some time depending on how close you are to the boiling time altitude threshold.

kylecorry31, 2021-05-01:
Note to put more details into the weather prediction, how the pressure in TS compares to weather stations, and how TS uses pressure data to create predictions.

kylecorry31, 2023-11-24:
Closing this in favor of #2074

closed 2023-11-24 by kylecorry31