kylecorry31/Trail-Sense #248: Decimal separator changed

aw-bib, 2020-10-18:
Something happened in 0.9.2 to the decimal separator "at some places". E.g. for the barometer it shows `1013,6 hPa` and `0,7 hPa/3h` in the app as well as in the notification. Also in some settings (e.g. `Visible beacon radius`) I get a comma instead of the dot. An explanation could be local formats: living in Germany people actually use a a comma instead of a dot as decimal marker. I just wonder how it comes about as my phone speaks English all the time and usually uses _the real thing_ in all other apps. It seems only to happen to Trail Sense.

kylecorry31, 2020-10-18:
I will look into this, I added the decimal formatting to the strings.xml file - it must be due to system locale. Would you prefer the period over the comma?

kylecorry31, 2020-10-18:
From where I found by searching online quick is changing their device language to US English would cause the decimal point to appear, but another variant of English may cause the comma to appear

aw-bib, 2020-10-19:
> Would you prefer the period over the comma?

I admit, being completely non-German here and indeed always write a decimal dot for years.

> From what I found by searching online quick is changing their device language to US English would cause the decimal point to appear, but another variant of English may cause the comma to appear

You're right, indeed. It was set to `English (Germany)` (whatever this is referring to.) Thanks for the pointer. :)

closed 2020-10-19 by aw-bib